prusa3d/PrusaSlicer #12880: Sign in to Prusa Connect doesn’t autofill or work with Sign in with Apple

bnewendorp, 2024-06-22:
### Description of the bug

In 2.8.0-beta1, I was not able to use Autofill from my iCloud Keychain to enter my credentials for PrusaConnect, nor could I use a keyboard expansion shortcut to enter text. Similarly Sign in with Apple did not work.

### Project file & How to reproduce

1. Have Prusa credentials in iCloud Keychain
2. Try signing in
3. No Autofill is suggested

For Sign in with Apple
1. Have Sign in with Apple set up for your Prusa account
2. Try using it within PrusaSlicer
3. Nothing happens

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0-beta1

### Operating system

macOS Sequoia beta 1

### Printer model

MK4

JakobGlieder, 2024-06-28:
still in the final release. that's a bad security flaw.

arekm, 2024-07-29:
... or with any password manager (but it was working fine before 2.8 alpha 5).

closed 2026-09-29 by prusaslicer-bot[bot]