prusa3d/PrusaSlicer #12881: PrusaSlicer 2.8.0-beta1 crash in preview

MarkusReichart, 2024-06-22:
### Description of the bug

PrusaSlicer just crash selecting "Color Print" as feature type. 

[image]

### Project file & How to reproduce

Open PS. Add a box shape. Auto Slice. Switch to preview. Select "Color Print" as feature type in the legend.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0-beta1

### Operating system

Windows 11 Home 23H2

### Printer model

Prusa MK3.5 MMU, Prusa MINI+

lukasmatena, 2024-06-24:
Please, enclose a 3MF. We are struggling to reproduce the issue. Thanks.

Jan-Soustruznik, 2024-06-24:
Hello, @MarkusReichart, thank you for your report. We found where should be problem. We will try to fix in some of next release. Have a nice day.

issue created: SPE-2369

lukasmatena, 2024-06-25:
This should be fixed in [2.8.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0-rc1). Closing.

closed 2024-06-25 by lukasmatena