prusa3d/PrusaSlicer #12884: 2.8.0 beta1 issues with building

fordfrog, 2024-06-22:
### Description of the bug

on gentoo linux (with boost-1.85.0) there are some missing includes that prevent compilation. we also had problem with linking prusaslicer because of missing libcrypto. here are the two patches that we use to evade those issues (i'm not sure they're correct though):
https://github.com/gentoo/gentoo/blob/master/media-gfx/prusaslicer/files/prusaslicer-2.8.0-missing-includes.patch
https://github.com/gentoo/gentoo/blob/master/media-gfx/prusaslicer/files/prusaslicer-2.8.0-fixed-linking.patch

### Project file & How to reproduce

not related

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0 beta1

### Operating system

gentoo linux

### Printer model

not related

fordfrog, 2024-06-25:
i updated the includes for r1. the links in the original post are still valid.

mogorman, 2024-06-28:
I had to patch the 2.8.0 release the same way  for nixos
```
--- a/src/slic3r/CMakeLists.txt
+++ b/src/slic3r/CMakeLists.txt
@@ -376,6 +376,7 @@ set(SLIC3R_GUI_SOURCES
 )
 
 find_package(NanoSVG REQUIRED)
+find_package(OpenSSL REQUIRED)
 
 if (APPLE)
     list(APPEND SLIC3R_GUI_SOURCES
@@ -404,7 +405,7 @@ endforeach()
 
 encoding_check(libslic3r_gui)
 
-target_link_libraries(libslic3r_gui libslic3r avrdude libcereal imgui libvgcode GLEW::GLEW OpenGL::GL hidapi libcurl ${wxWidgets_LIBRARIES} NanoSVG::nanosvg NanoSVG::nanosvgrast)
+target_link_libraries(libslic3r_gui libslic3r avrdude libcereal imgui libvgcode GLEW::GLEW OpenGL::GL hidapi libcurl ${wxWidgets_LIBRARIES} NanoSVG::nanosvg NanoSVG::nanosvgrast OpenSSL::SSL OpenSSL::Crypto)
 
 if (MSVC)
     target_link_libraries(libslic3r_gui Setupapi.lib)
```

amatulic, 2024-07-30:
Might be related to #13037 on Windows

adelton, 2025-01-01:
Just for the record, in Fedora I just used `-DCMAKE_EXE_LINKER_FLAGS=-lcrypto` to link with the system OpenSSL: https://src.fedoraproject.org/rpms/prusa-slicer/c/28e721f618228f167db11af1cfe65d3bcb337c87?branch=rawhide

amatulic, 2026-09-02:
Issue resolved itself for me around version 2.9.4. @fordfrog maybe you can close it.