prusa3d/PrusaSlicer #12890: [Multi material] Support interfece material is used in the main support structure, causes tree support breakages

viesturz, 2024-06-23:
### Description of the bug

Tree like supports are breaking apart when using a support material for support interface only.

Support material adhesion is usually significantly lower than the main material.
Think PLA+PETG, HIPS+ABS. This makes nice snapp-off supports.

Only specifying support material for support interfaces makes for less toll changes faster prints.
However PrusaSlicer sprinkles in the support material into other non-interface supports at the same layer.
Which wreaks havoc for tree supports that need significant rigidity during print.

**Describe the solution you'd like**
Only apply this setting to support interfaces, leave supports with the main filament.
This would not affect print speed, as each layer typically has at least one main filament being printed.

**Additional context**
Add any other context or paint drawing about the feature request here.

### Project file & How to reproduce

[image]
[Mount Square Rods.zip](https://github.com/user-attachments/files/15943969/Mount.Square.Rods.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1; 2.8 beta

### Operating system

Win11

### Printer model

voron 2.4 + toolchanger

viesturz, 2024-06-23:
Orca slicer does this with support_interface_not_for_body option.

jhnphm, 2024-12-16:
I've also observed issues with this happening on the wipe tower

closed 2026-09-29 by prusaslicer-bot[bot]