prusa3d/PrusaSlicer #12891: Could you add alignment options for object alignment?

cihadkocasahan, 2024-06-23:
**Is your feature request related to a problem? Please describe.**
I typically utilize the automatic arrange function, although there are occasions when I prefer to manually arrange objects. However, I encounter difficulties with alignment during manual arrangements. Could alignment options, such as aligning left edges, right edges, centering horizontally, centering vertically, or any other alignment options you can suggest, be incorporated under the 'arrange' feature available in most IDEs to address this issue?

**Describe the solution you'd like**
[image]

[image]

cihadkocasahan, 2024-06-23:
More complex one :
[image]

HonzaJaros, 2024-06-30:
would be great if this function is also for added objects, to align in centre. jet say you want to put hole onto your model as a negative cylinder, but you also want to add a but of material around the hole so you put positive cylinder. now would be nice if we can align those 2 cylinders to the centre of each other.

same with text for example, align to centre and bottom of the object. click....

Tupson444, 2026-09-19:
still wanted

Jan-Soustruznik, 2026-10-21:
A similar feature request already exists. This request has been merge to keep topics organized.

closed 2026-10-21 by Jan-Soustruznik