prusa3d/PrusaSlicer #12894: Mouse cursor always in loading state after open "Physical Printer"

shinanca, 2024-06-23:
### Description of the bug

When open tab "Physical Printer", then go to any tab or stay on web interface mouse icon always in loading state.

### Project file & How to reproduce

Prusa Slicer-> Physical Printer then go to any tab. If don`t open "Physical Printer" tab then no problem at all.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0-beta1

### Operating system

Linux

### Printer model

Flying Bear

cocasema, 2024-06-25:
I can confirm the same/similar issue with Voron + Mainsail. The Mainsail loading animation freezes after a few seconds, and after LMC the background goes blank. 

https://github.com/prusa3d/PrusaSlicer/assets/9308400/0a0f822f-d27f-46ed-96e0-9468ec5ce4bd

lunatic1972, 2024-06-28:
Mine does the same thing and I'm on Manjaro latest and using latest Prusa Slicer (2.8.0). You click on the physical printer tab and the cursor never changes back from there. It does however still seem to function on mine where I can still use the software but just kind of an annoying bug where the cursor never changes back to an arrow like it should.

timur-tabi, 2024-08-18:
I notice this also with Ubuntu 22.04 and PrusaSlicer 2.8.0.

SamChill, 2024-09-24:
I still see this on PrusaSlicer 2.8.1 on Fedora 40 using KDE.

itsByte, 2024-11-18:
Still happening on 2.8.1, Flatpak on Fedora 41 GNOME

dgrant, 2024-11-21:
For me, I'm using Octopi and this is an issue. Note: I have to actually log-in to Octopi when the login screen comes up, to reproduce this. I always do need to log in though, just mentioning this in case anyone is trying to repro this.

timur-tabi, 2024-11-24:
Still happening in 2.9.0-alpha1.  In fact, the problem is worse now because it now happens in both the "Printables" and "Physical Printer" tabs.

mw75, 2024-12-05:
I disabled connect, printables and printer webview in MainFrame.cpp of Version 2.9.0-alpha1, but somehow the mouse pointer got set to the spinner anyway. Dropping appimages forced me to compile myself and that allows me do such modifications....

timur-tabi, 2025-01-22:
Using 2.9.0 and Ubuntu 24, I no longer see this issue.

Now this it works properly, this is a really neat feature.  I used to have to search all of my browser tabs/windows across four different workspaces every time I needed Octoprint.  Not any more!

closed 2026-09-29 by prusaslicer-bot[bot]