prusa3d/PrusaSlicer #12895: 2.8.0. beta-1 - None of the top-level tabs work (macOS 14.5 ARM64)

spkane, 2024-06-23:
### Description of the bug

I can not switch to settings of the plater using the new tabs, only the old menu entries work properly.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0-beta1

### Operating system

macOS 14.5

### Printer model

MK4

acpigeon, 2024-06-27:
Also having this issue with the stable release (2.8.0). It is unusable in this state.
MacOS 14.5
MK3S+

spkane, 2024-06-29:
I am no longer seeing this issue in 2.8.0.

@acpigeon Are you still seeing it with a new download of the application?

acpigeon, 2024-06-29:
I did a reinstall and still saw the same issue. I also had a bunch of other issues, e.g. the workplace not being able to be rotated and inability to select objects or manipulate them. I downgraded back to 2.7.5-rc1 rather than mess with things further.

closed 2024-08-13 by spkane