prusa3d/PrusaSlicer #12896: 1.8.0-beta1 - No bitmap handler for type 15 defined

spkane, 2024-06-23:
### Description of the bug

I was trying to open the "About Prusa" dialog and suddenly for this message:

[image]

Maybe related to this setting? As the UI seemed to open up the prefs when this error occurred, but I closed it before I noted the exact spot.

[image]

### Project file & How to reproduce

?

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0-beta1

### Operating system

macos 14.5 ARM64

### Printer model

MK4

lukasmatena, 2024-06-24:
Can you please post the file that it complains about? Thanks.

JitaoNeveitt, 2024-06-27:
I too have this bug. whenever I go into the configuration tab, it shows the same message...when I hit "ok" the app closes.  Also, this may be unrelated, but I have noticed that when I log in to my Prusa account through the slicer, the app doesn't really respond.  (PrusaSlicer 2.8.0 Final)
[image]

spkane, 2024-06-27:
> Can you please post the file that it complains about? Thanks.

@lukasmatena The file is not on the system anymore and honestly I have no idea what that file is. If I see this again (I went back to the old version) I'll try and grab it.

What exactly is the `prusaslicer-avatar-*.png` image for?

Maybe my cloud user login image?

If so, maybe this is a corrupted or impartial download. If so, handling this error gracefully would be good, since it is likely not critical to the application functionality.

spkane, 2024-06-28:
I can confirm with the official release that opening up the configuration wizard causes this to happen and then fully crashes the application, so I can't really see the whole message.

spkane, 2024-06-28:
@lukasmatena This is the file.

`/var/folders/rd/9w38b2m139d3tfvp186pkt180000gn/T/prusaslicer-avatar-1840334539247780131.png`

[image]

And it is a JPG in reality, so this is likely the issue.

```
$ file /var/folders/rd/9w38b2m139d3tfvp186pkt180000gn/T/prusaslicer-avatar-1840334539247780131.png

/var/folders/rd/9w38b2m139d3tfvp186pkt180000gn/T/prusaslicer-avatar-1840334539247780131.png: JPEG image data, JFIF standard 1.01, aspect ratio, density 1x1, segment length 16, Exif Standard: [TIFF image data, little-endian, direntries=1, software=Google], baseline, precision 8, 96x96, components 3
```

jacobwtyler, 2024-06-28:
same issue

jacobwtyler, 2024-06-28:
I went to connect.prusa3d.com and changed my profile picture (auto uploaded by SSO with google) and that seemed to fix the error.

spkane, 2024-06-28:
> I went to connect.prusa3d.com and changed my profile picture (auto uploaded by SSO with google) and that seemed to fix the error.

It is good to know that there is a reasonable workaround for the time being until the underlying issue gets fixed.

lukasmatena, 2024-06-28:
Thanks. We managed to find and fix the problem. It was not in PrusaSlicer, but on the server side. The fix was already deployed, so this issue should no longer show in any PrusaSlicer version.

@spkane 

> And it is a JPG in reality, so this is likely the issue.

It was.

Avatars from Prusa Account showed correctly, the problem appeared when avatar from Google account was used.

Can you guys confirm that the issue really disappeared?
Duplicate of #12860.

spkane, 2024-06-29:
It appears as if this is fixed. Thank you for the quick turn around!

closed 2024-06-29 by spkane