prusa3d/PrusaSlicer #12903: Crashing when slicing big models but now if they are sized down

branden5002, 2024-06-24:
### Description of the bug

It crashes when I try to slice big models this happens with the helmet as well as the other body parts besides the small pieces like the hand armor and that. it normally would crash with no error message.

### Project file & How to reproduce

[Chest.zip](https://github.com/user-attachments/files/15961092/Chest.zip)
to reproduce just click slice now and it crashes

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.7+win64

### Operating system

Windows11

### Printer model

Elegoo Neptune 3 max

branden5002, 2024-06-24:
I also tried the beta

Jan-Soustruznik, 2024-06-25:
Hello, @branden5002, thank you for your report, but we cannot reproduce your reported problem. We can slice this model without problem. Could you share more info, please? HW spec of your PC, what App was running with slicer, etc .. Thank you.

branden5002, 2024-06-25:
@Jan-Soustruznik ive never sent spec before but I think i got it?
[image]
[image]

closed 2026-09-29 by prusaslicer-bot[bot]