prusa3d/PrusaSlicer #12905: Vulnerabilities in libigl: CVE-2024-24686 CVE-2024-24685 CVE-2024-24684 CVE-2024-24584 CVE-2024-24583 CVE-2024-23951 CVE-2024-23950 CVE-2024-23949 CVE-2024-23948 CVE-2024-23947 CVE-2024-22181 CVE-2023-49600 CVE-2023-35953 CVE-2023-35952 CVE-2023-35951 CVE-2023-35950 CVE-2023-35949

hyperair, 2024-06-25:
### Description of the bug

The following vulnerabilities have just been published for libigl, which PrusaSlicer embeds a copy of:

https://talosintelligence.com/vulnerability_reports/TALOS-2024-1929
https://talosintelligence.com/vulnerability_reports/TALOS-2024-1928
https://talosintelligence.com/vulnerability_reports/TALOS-2024-1926
https://talosintelligence.com/vulnerability_reports/TALOS-2024-1930
https://talosintelligence.com/vulnerability_reports/TALOS-2023-1879
https://talosintelligence.com/vulnerability_reports/TALOS-2023-1784

https://github.com/libigl/libigl/issues/2387

Forwarded from https://bugs.debian.org/1074233

libigl 2.4.0 and 2.5.0 are affected. Not sure what version PrusaSlicer embeds.

CVE-2024-24686[0]:
| Multiple stack-based buffer overflow vulnerabilities exist in the
| readOFF functionality of libigl v2.5.0. A specially crafted .off
| file can lead to stack-based buffer overflow. An attacker can
| provide a malicious file to trigger this vulnerability.This
| vulnerability concerns the parsing of comments within the faces
| section of an `.off`  file processed via the `readOFF` function.

CVE-2024-24685[1]:
| Multiple stack-based buffer overflow vulnerabilities exist in the
| readOFF functionality of libigl v2.5.0. A specially crafted .off
| file can lead to stack-based buffer overflow. An attacker can
| provide a malicious file to trigger this vulnerability.This
| vulnerability concerns the parsing of comments within the vertex
| section of an `.off`  file processed via the `readOFF` function.

CVE-2024-24684[2]:
| Multiple stack-based buffer overflow vulnerabilities exist in the
| readOFF functionality of libigl v2.5.0. A specially crafted .off
| file can lead to stack-based buffer overflow. An attacker can
| provide a malicious file to trigger this vulnerability.This
| vulnerability concerns the header parsing occuring while processing
| an `.off`  file via the `readOFF` function.       We can see above
| that at [0] a stack-based buffer called `comment` is defined with an
| hardcoded size of `1000 bytes`.  The call to `fscanf` at [1] is
| unsafe and if the first line of the header of the `.off` files is
| longer than 1000 bytes it will overflow the `header` buffer.

CVE-2024-24584[3]:
| Multiple out-of-bounds read vulnerabilities exist in the readMSH
| functionality of libigl v2.5.0. A specially crafted .msh file can
| lead to an out-of-bounds read. An attacker can provide a malicious
| file to trigger this vulnerability.This vulnerabilitty concerns
| the`readMSH` function while processing `MshLoader::ELEMENT_TET`
| elements.

CVE-2024-24583[4]:
| Multiple out-of-bounds read vulnerabilities exist in the readMSH
| functionality of libigl v2.5.0. A specially crafted .msh file can
| lead to an out-of-bounds read. An attacker can provide a malicious
| file to trigger this vulnerability.This vulnerabilitty concerns
| the`readMSH` function while processing `MshLoader::ELEMENT_TRI`
| elements.

CVE-2024-23951[5]:
| Multiple improper array index validation vulnerabilities exist in
| the readMSH functionality of libigl v2.5.0. A specially crafted .msh
| file can lead to an out-of-bounds write. An attacker can provide a
| malicious file to trigger this vulnerability.This vulnerability
| concerns the `igl::MshLoader::parse_element_field` function while
| handling an `ascii`.msh` file.

CVE-2024-23950[6]:
| Multiple improper array index validation vulnerabilities exist in
| the readMSH functionality of libigl v2.5.0. A specially crafted .msh
| file can lead to an out-of-bounds write. An attacker can provide a
| malicious file to trigger this vulnerability.This vulnerability
| concerns the `igl::MshLoader::parse_element_field` function while
| handling an `binary`.msh` file.

CVE-2024-23949[7]:
| Multiple improper array index validation vulnerabilities exist in
| the readMSH functionality of libigl v2.5.0. A specially crafted .msh
| file can lead to an out-of-bounds write. An attacker can provide a
| malicious file to trigger this vulnerability.This vulnerability
| concerns the `igl::MshLoader::parse_node_field` function while
| handling an `ascii`.msh` file.

CVE-2024-23948[8]:
| Multiple improper array index validation vulnerabilities exist in
| the readMSH functionality of libigl v2.5.0. A specially crafted .msh
| file can lead to an out-of-bounds write. An attacker can provide a
| malicious file to trigger this vulnerability.This vulnerability
| concerns the `igl::MshLoader::parse_nodes` function while handling
| an `ascii`.msh` file.

CVE-2024-23947[9]:
| Multiple improper array index validation vulnerabilities exist in
| the readMSH functionality of libigl v2.5.0. A specially crafted .msh
| file can lead to an out-of-bounds write. An attacker can provide a
| malicious file to trigger this vulnerability.This vulnerability
| concerns the `igl::MshLoader::parse_nodes` function while handling a
| `binary` `.msh` file.

CVE-2024-22181[10]:
| An out-of-bounds write vulnerability exists in the readNODE
| functionality of libigl v2.5.0. A specially crafted .node file can
| lead to an out-of-bounds write. An attacker can provide a malicious
| file to trigger this vulnerability.

CVE-2023-49600[11]:
| An out-of-bounds write vulnerability exists in the PlyFile
| ply_cast_ascii functionality of libigl v2.5.0. A specially crafted
| .ply file can lead to a heap buffer overflow. An attacker can
| provide a malicious file to trigger this vulnerability.

CVE-2023-35953[12]:
| Multiple stack-based buffer overflow vulnerabilities exist in the
| readOFF.cpp functionality of libigl v2.4.0. A specially-crafted .off
| file can lead to a buffer overflow. An attacker can arbitrary code
| execution to trigger these vulnerabilities.This vulnerability exists
| within the code responsible for parsing comments within the
| geometric vertices section within an OFF file.

CVE-2023-35952[13]:
| Multiple stack-based buffer overflow vulnerabilities exist in the
| readOFF.cpp functionality of libigl v2.4.0. A specially-crafted .off
| file can lead to a buffer overflow. An attacker can arbitrary code
| execution to trigger these vulnerabilities.This vulnerability exists
| within the code responsible for parsing comments within the
| geometric faces section within an OFF file.

CVE-2023-35951[14]:
| Multiple stack-based buffer overflow vulnerabilities exist in the
| readOFF.cpp functionality of libigl v2.4.0. A specially-crafted .off
| file can lead to a buffer overflow. An attacker can arbitrary code
| execution to trigger these vulnerabilities.This vulnerability exists
| within the code responsible for parsing geometric vertices of an OFF
| file.

CVE-2023-35950[15]:
| Multiple stack-based buffer overflow vulnerabilities exist in the
| readOFF.cpp functionality of libigl v2.4.0. A specially-crafted .off
| file can lead to a buffer overflow. An attacker can arbitrary code
| execution to trigger these vulnerabilities.This vulnerability exists
| within the code responsible for parsing the header of an OFF file.

CVE-2023-35949[16]:
| Multiple stack-based buffer overflow vulnerabilities exist in the
| readOFF.cpp functionality of libigl v2.4.0. A specially-crafted .off
| file can lead to a buffer overflow. An attacker can arbitrary code
| execution to trigger these vulnerabilities.This vulnerability exists
| within the code responsible for parsing geometric faces of an OFF
| file.

### Project file & How to reproduce

N/A

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

unknown

### Operating system

All

### Printer model

N/A

lukasmatena, 2024-06-25:
Thanks for the info. PrusaSlicer should not be affected by any of these. While we indeed bundle libigl and use it in a couple of places, we do not use it to read any files. All the vulnerabilities above point to functions handling various file formats.

hyperair, 2024-06-26:
Thanks for the confirmation. I guess we can close this bug then

closed 2024-06-28 by hyperair