prusa3d/PrusaSlicer #12907: Unable read avatar from account

RaySlovakia, 2024-06-25:
### Description of the bug

Unable read avatar png from account after login Prusa account

[image]

### Project file & How to reproduce

None

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0-rc1

### Operating system

Windows 11

### Printer model

Anycubic Kobra 2 Pro

Jan-Soustruznik, 2024-06-25:
https://github.com/prusa3d/PrusaSlicer/issues/12860 closed as duplicate

closed 2024-06-25 by Jan-Soustruznik

lukasmatena, 2024-06-28:
Thanks. As described in https://github.com/prusa3d/PrusaSlicer/issues/12860#issuecomment-2196207417, we managed to find and fix the problem. It was not in PrusaSlicer, but on the server side. The fix was already deployed, so this issue should no longer show in any PrusaSlicer version.

Avatars from Prusa Account showed correctly, the problem appeared when avatar from Google account was used.

Can you confirm that the issue really disappeared?

RaySlovakia, 2024-06-28:
> Thanks. As described in #12860 (comment)

Thx for info
Yes, now show avatar correctly, problem solved
Thank You 😃