prusa3d/PrusaSlicer #12908: PrusaSlicer 2.8.0 rc1 PREVIEW empty

Kito221170, 2024-06-25:
### Description of the bug

PREVIEW is empty after slicing.(2.8.0 rc1)
2.7.4 is ok.
[image]
[image]
[image]
[image]

### Project file & How to reproduce

[image]
[image]
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.0-rc1+win64-202406242049

### Operating system

Edição	Windows 11 Pro Versão	22H2 Instalado em	‎16/‎06/‎2024 Compilação do SO	22621.3737 Experiência	Windows Feature Experience Pack 1000.22700.1009.0

### Printer model

Elegoo Neptune 2S, Creality Ender 3

Golfsyn, 2024-06-26:
Also an empty preview

TerPuc, 2024-06-26:
Hi @Kito221170, thank you for you report. Unfortunately I was not able to reproduce this issue. May I kindly ask you to provide me with 3mf of the project which shows this behavior for further investigation? Thank you and have a nice day!

TerPuc, 2024-06-27:
Hi again! Yes, I do understand that. However, we will need to have a look at your settings and configurations that might be causing this. That is why I would like to ask you to save one of the models as 3MF and add it to this issue. Thank you!

Kito221170, 2024-06-27:
3MF (https://drive.google.com/drive/folders/1RXTUumMreS2LNV93slJmbJlVy-fETcUb?usp=drive_link)

Kito221170, 2024-06-27:
[PreviewEmpty.zip](https://github.com/user-attachments/files/16019570/PreviewEmpty.zip)

Megastrong1, 2024-06-27:
I have the same problem
[image]
[Держатель экрана.zip](https://github.com/user-attachments/files/16021129/default.zip)

Kito221170, 2024-06-27:
The problem continues with the final version 2.8.0.

OneTouch3D, 2024-06-28:
I'm having the same problem.
After updating to Prusa Version 2.8.0+win64, the system when slicing the part does not appear on the sliced ​​table, the menus are all ok, just the part disappears, I can use any standard printer in Prusa and it happens the same.
Windows 10 Pro
[image]

TerPuc, 2024-06-28:
Thank you all for submiting the files. But we still couldn't reproduce. It might be connected to OpenGL. Would anyone be so kind to share the system info screen (Help->System Info)? Thanks!

pisuki, 2024-06-28:
> Thank you all for submiting the files. But we still couldn't reproduce. It might be connected to OpenGL. Would anyone be so kind to share the system info screen (Help->System Info)? Thanks!

I´m having the same problem right now and try to install newest grafic driver right now. 

PrusaSlicer
Version:   2.8.0+win64
Build:     PrusaSlicer-2.8.0+win64-202406270929

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (Erzeugungsversion 19045), 64-bit Edition
Total RAM size [MB]: 34,256MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 552.44)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 2070 SUPER/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
[... 328 lines cut ...]
GL_NV_transform_feedback2
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

pisuki, 2024-06-28:
Ok, installing new grafic driver didn´t change anything.

PrusaSlicer
Version:   2.8.0+win64
Build:     PrusaSlicer-2.8.0+win64-202406270929

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (Erzeugungsversion 19045), 64-bit Edition
Total RAM size [MB]: 34,256MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 556.12)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 2070 SUPER/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
[... 327 lines cut ...]
GL_NV_transform_feedback2
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

Rob70ar, 2024-06-28:
Same problem here.

PrusaSlicer
Version:   2.8.0+win64
Build:     PrusaSlicer-2.8.0+win64-202406270929

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (compilación 19045), edición de 64 bits
Total RAM size [MB]: 8,570MB
OpenGL installation
GL version:   3.3.0 (3.3.0)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     GeForce 315M/PCIe/SSE2
GLSL version: 3.30.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_clear_buffer_object
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
[... 173 lines cut ...]
GL_NV_texture_expand_normal
GL_NV_texture_multisample
GL_NV_texture_rectangle
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

Kito221170, 2024-06-28:
Nome do Sistema Operacional	Microsoft Windows 11 Pro
Versão	10.0.22621 Compilação 22621
Outra Descrição do Sistema Operacional 	Não disponível
Fabricante do Sistema Operacional	Microsoft Corporation
Nome do sistema	SINGDESKTOP
Fabricante do sistema	System manufacturer
Modelo do sistema	System Product Name
Tipo do sistema	PC baseado em X64
SKU do sistema	To Be Filled By O.E.M.
Processador	AMD FX(tm)-6300 Six-Core Processor, 3500 Mhz, 3 Núcleo(s), 6 Processador(es) Lógico(s)
Versão/data do BIOS	American Megatrends Inc. 1201, 04/03/2017
Versão do SMBIOS	2.5
Versão do Controlador Incorporado	255.255
Modo da BIOS	Herdado
Fabricante da BaseBoard	ASUSTeK Computer INC.
Produto BaseBoard	M5A78L-M LX/BR
Versão da BaseBoard	Rev X.0x
Função da Plataforma	Área de Trabalho
Estado da Inicialização Segura	Sem suporte
Configuração PCR7	Ligação Impossível
Pasta do Windows	C:\Windows
Pasta do sistema	C:\Windows\system32
Dispositivo de inicialização	\Device\HarddiskVolume1
Localidade	Brasil
Camada de Abstração de Hardware	Versão = "10.0.22621.2506"
Nome de usuário	SingDesktop\singd
Fuso horário	Hora oficial do Brasil
Memória Física (RAM) Instalada	8,00 GB
Memória física total	7,73 GB
Memória física disponível	3,05 GB
Memória virtual total	11,0 GB
Memória virtual disponível	5,60 GB
Espaço do arquivo de paginação	3,25 GB
Arquivo de paginação	C:\pagefile.sys
Proteção de DMA de Kernel	Desativado
Segurança baseada em virtualização	Não habilitado
Política do Controle de Aplicativos do Windows Defender	Imposto
Política do modo de usuário do Controle de Aplicativos do Windows Defender	Desativado
Suporte à Criptografia de Dispositivo	Motivos da falha da criptografia de dispositivo automática: TPM não utilizável, Associação PCR7 não compatível, Falha na Interface de Teste de Segurança de Hardware, e o dispositivo não está em Espera Moderna, Barramento/dispositivos compatíveis com DMA não permitidos detectados, TPM não utilizável
Hyper-V - Extensões de Modo Monitor VM	Sim
Hyper-V - Extensões de Conversão de Endereços de Segundo Nível	Sim
Hyper-V - Virtualização Habilitada no Firmware	Sim
Hyper-V - Proteção de Execução de Dados	Sim

Nome	NVIDIA GeForce 210
Identificação de dispositivo PNP	PCI\VEN_10DE&DEV_0A65&SUBSYS_80941462&REV_A2\4&2A1FC852&0&0010
Tipo de adaptador	GeForce 210, compatível com NVIDIA
Descrição do adaptador	NVIDIA GeForce 210
Memória RAM do adaptador	1,00 GB (1.073.741.824 bytes)
Drivers instalados	nvd3dumx.dll,nvwgf2umx.dll,nvwgf2umx.dll
Versão do driver	21.21.13.4201
Arquivo INF	oem15.inf (seção Section004)
Planos de cores	Não disponível
Entradas da tabela de cores	4294967296
Resolução	1920 x 1080 x 60 Hertz
Bits/Pixel	32
Endereço de memória	0xFD000000-0xFDFFFFFF
Endereço de memória	0xD0000000-0xDFFFFFFF
Endereço de memória	0xCE000000-0xCFFFFFFF
Porta de E/S	0x0000DC80-0x0000DCFF
Canal IRQ	IRQ 18
Driver	C:\WINDOWS\SYSTEM32\DRIVERS\NVLDDMKM.SYS (21.21.13.4201, 12,32 MB (12.914.360 bytes), 16/06/2024 19:54)
	
Nome	ATI Radeon 3000 Graphics
Identificação de dispositivo PNP	PCI\VEN_1002&DEV_9616&SUBSYS_83881043&REV_00\4&52E5CC6&0&2808
Tipo de adaptador	ATI display adapter (0x9616), compatível com Advanced Micro Devices, Inc.
Descrição do adaptador	ATI Radeon 3000 Graphics
Memória RAM do adaptador	256,00 MB (268.435.456 bytes)
Drivers instalados	aticfx64.dll,aticfx64.dll
Versão do driver	8.970.100.9001
Arquivo INF	oem6.inf (seção ati2mtag_RS780)
Planos de cores	Não disponível
Entradas da tabela de cores	4294967296
Resolução	1280 x 1024 x 60 Hertz
Bits/Pixel	32
Endereço de memória	0xB0000000-0xBFFFFFFF
Porta de E/S	0x0000C000-0x0000CFFF
Endereço de memória	0xFCFF0000-0xFCFFFFFF
Endereço de memória	0xFCE00000-0xFCFFFFFF
Canal IRQ	IRQ 4294967293
Porta de E/S	0x000003B0-0x000003BB
Porta de E/S	0x000003C0-0x000003DF
Endereço de memória	0xA0000-0xBFFFF
Driver	C:\WINDOWS\SYSTEM32\DRIVERS\ATIKMPAG.SYS (8.14.1.6264, 351,50 KB (359.936 bytes), 13/01/2015 16:20)

Golfsyn, 2024-06-28:
PrusaSlicer
Version:   2.8.0+win64
Build:     PrusaSlicer-2.8.0+win64-202406270929

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (сборка 19045), 64-бит редакция
Total RAM size [MB]: 4,293MB
OpenGL installation
GL version:   3.3.0 (3.3.0)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     GeForce 9600 GT/PCIe/SSE2
GLSL version: 3.30.0
Textures compression:       Enabled
Mutisampling: Enabled (0 samples)
<details>
<summary>Installed extensions:</summary>
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_clear_buffer_object
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robust_buffer_access_behavior
GL_ARB_robustness
[... 161 lines cut ...]
GL_NV_texture_env_combine4
GL_NV_texture_expand_normal
GL_NV_texture_multisample
GL_NV_texture_rectangle
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

malowz, 2024-06-28:
mine:

PrusaSlicer
Version:   2.7.5-rc1+win64
Build:     PrusaSlicer-2.7.5-rc1+win64-202406070441

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 7 (build 7601, Service Pack 1), 64-bit edition
Total RAM size [MB]: 6,441MB
OpenGL installation
GL version:   3.3.0 (3.3.0)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     GeForce 210/PCIe/SSE2
GLSL version: 3.30.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_clear_buffer_object
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
[... 172 lines cut ...]
GL_NV_texture_expand_normal
GL_NV_texture_multisample
GL_NV_texture_rectangle
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

malowz, 2024-06-28:
info: gcodeviewer also do not show the part on the bed from the generated (properly sliced) gcode

arshakuni, 2024-06-28:
Same issue, came right after updating slicer to 2.8.0.
Nothing helped from the following. restart program/windows/graphic driver, tested old working projects, etc...

It is interesting that the gcode implementation is working and even nozzle model appears visible while working with sliders
[image]

arshakuni, 2024-06-28:
> Same issue, came right after updating slicer to 2.8.0. Nothing helped from the following. restart program/windows/graphic driver, tested old working projects, etc...
> 
> It is interesting that the gcode implementation is working and even nozzle model appears visible while working with sliders ![2024-06-28_20-41-31](https://private-user-images.githubusercontent.com/36967476/344248013-09a0c4a3-1d27-4dd2-a9a1-2fc19e50b1fb.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk1OTMzNTAsIm5iZiI6MTcxOTU5MzA1MCwicGF0aCI6Ii8zNjk2NzQ3Ni8zNDQyNDgwMTMtMDlhMGM0YTMtMWQyNy00ZGQyLWE5YTEtMmZjMTllNTBiMWZiLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI4VDE2NDQxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTllOTVjOGZkYjRmYTA2ODgzODlmM2Y3NTJhZjdjYzM3N2Y4YTQ5MmRjOTA1ZTAwYWMwYzAxZTM3NmIzMjJiZGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.lT-0VlZwgf3SPvTCf_neFuHd8iQ3ASs5_tfD2i3EjSM)

https://github.com/prusa3d/PrusaSlicer/issues/12947

arshakuni, 2024-06-28:
> info: gcodeviewer also do not show the part on the bed from the generated (properly sliced) gcode

same as mine

malowz, 2024-06-28:
**NEWS**: using the **--sw-renderer** on commandline, the sliced is properly shown.

so, the problem is most likely on the OpenGL as [TerPuc](https://github.com/TerPuc) mentioned.

Kito221170, 2024-06-28:
@Malowz. Ok
prusa-slicer-console.exe --sw-renderer ok

Rob70ar, 2024-07-01:
> **NEWS**: using the **--sw-renderer** switch on commandline, the sliced part is properly shown.
tried this and also worked on my computer

gmarconi2, 2024-07-01:
Scusate come si usa l' opzione  --sw-renderer sulla riga di comando.
Scrivere ho provato questo e ha funzionato anche sul mio computer è come dire ho riparato la mia auto che non funzionava, e ora funziona, di come hai fatto in modo che possiamo farlo anche noi. !!!

Rob70ar, 2024-07-01:
> Scusate come si usa l' opzione --sw-renderer sulla riga di comando. Scrivere ho provato questo e ha funzionato anche sul mio computer è come dire ho riparato la mia auto che non funzionava, e ora funziona, di come hai fatto in modo che possiamo farlo anche noi. !!!

You need to launch the Slicer from Windows command line: launch CMD, then go to Prusa Slicer directory and type: PrusaSlicer -sw-renderer
Slicer will show the result, but will function terrible slow... useless..

fitimollas, 2024-07-02:
Same problem here.
W10Pro 22H2
NVIDIA GeForce GT 220

brigthub, 2024-07-03:
Same problem here.  PrusaSlicer 2.8.0, W10Pro, running in Vmware.   PrusaSlicer 2.7.1 works fine on same setup.  Latest OrcaSlicer also previews or renders gcode on same setup.

Golfsyn, 2024-07-03:
> Извините, если вы используете опцию --sw-renderer на странице команды. Я доказал это, и мой компьютер также функционировал, поэтому я решил, что моя машина не работает, а сейчас она работает так же, как мы. !!!

This method doesn't work for me, the preview is still empty

Jedrek76, 2024-07-03:
The same problem after update.

gmarconi2, 2024-07-03:
If you want to try version 2.8 try this batch file
Change the directory if yours is different.
[prusa2.8.txt](https://github.com/user-attachments/files/16089498/prusa2.8.txt)
Cambiate txt in bat

dwprod38, 2024-07-06:
New PrusaSlicer 2.8.0 had warning message after launching application " Error running JavaScript: TypeError: undefined is not an object (evaluating 'window.webkit.messageHandlers')" and impossible to view layers in "Preview" after slicing generation... (MacOS 10.12.6)

Back to Prusaslicer 2.7.4 and works fine...

[image]

[image]
[image]

brigthub, 2024-07-06:
> Same problem here. PrusaSlicer 2.8.0, W10Pro, running in Vmware. PrusaSlicer 2.7.1 works fine on same setup. Latest OrcaSlicer also previews or renders gcode on same setup.

2024-07-06 -- additional information.  The preview works if the stl model being sliced is only ONE layer tall.

Galinette404, 2024-07-06:
Hello,
I have the same issue.
It works by using --sw-renderer, but the soft is slow.
Here is my configuration : 

PrusaSlicer
Version:   2.8.0+win64
Build:     PrusaSlicer-2.8.0+win64-202406270929

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (version 19045), édition 64-bit
Total RAM size [MB]: 8,431MB
OpenGL installation
GL version:   3.3.0 (3.3.0)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     Quadro FX 2800M/PCIe/SSE2
GLSL version: 3.30.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_clear_buffer_object
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
[... 166 lines cut ...]
GL_NV_texture_env_combine4
GL_NV_texture_expand_normal
GL_NV_texture_multisample
GL_NV_texture_rectangle
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

UrsulaHertzog, 2024-07-09:
Exact same problem here, too. Upgraded to 2.8.0 ==> GCode viewer empty.

KarelVogel, 2024-07-10:
For the authors of the slicer:
You have a problem with openGL version support. With version 3.3 slicer 2.8.0 does not show the preview, with version 4.6 it does. You have too new installations of your computers and therefore the problem is not manifesting itself for you. I think it would be a good idea not to make these mistakes in slicer, because using older HW just for 3D printing is a good choice. Please fix the code, or indicate from which version of openGL slicer will work in the future, for many people on this planet this will mean spending money on new HW and I think this is not a good idea.
And just don't say "whatever you want, it's free software" :)
Vorel

DerKameramann, 2024-07-10:
Having the exact same problems with performance and empty preview with my old MAC with Nvidia Graphics.
Maybe not an OS related problem.

Rob70ar, 2024-07-10:
It also works on a PC with Win 10 and OpenGL 4.3

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus.www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

kubispe1, 2024-07-10:
Hi,
After we find a physical device in our company or around to reproduce the issue, we will try to fix it. Don't forget to copy out your HW parameters. Just to mention, we are in contact with our support.
Thanks for your cooperation.
Here is a out ticket: **SPE-2411**

JavierBlancoGlez, 2024-07-10:
PrusaSlicer
Version:   2.8.0+win64
Build:     PrusaSlicer-2.8.0+win64-202406270929

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (compilación 19045), edición de 64 bits
Total RAM size [MB]: 17,113MB
OpenGL installation
GL version:   3.3.0 (3.3.0)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     GeForce 210/PCIe/SSE2
GLSL version: 3.30.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_clear_buffer_object
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robust_buffer_access_behavior
[... 171 lines cut ...]
GL_NV_texture_expand_normal
GL_NV_texture_multisample
GL_NV_texture_rectangle
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

DerKameramann, 2024-07-11:
NOT properly working (preview table - no show issue, gui performance drop):
-------------------------------------------------------------------
PrusaSlicer
Version:   2.8.0+MacOS-x64
Build:     PrusaSlicer-2.8.0+MacOS-x64-202406270936

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Mojave Version 10.14.6 (Build 18G103)
Total RAM size [MB]: 8,590MB
OpenGL installation
GL version:   3.3.0 (3.3 NVIDIA-10.4.14 310.90.30.05b27)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GT 330M OpenGL Engine
GLSL version: 3.30.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_APPLE_client_storage
GL_APPLE_container_object_shareable
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_texture_range
GL_ARB_ES2_compatibility
GL_ARB_blend_func_extended
GL_ARB_draw_buffers_blend
GL_ARB_explicit_attrib_location
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_occlusion_query2
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shading_language_include
GL_ARB_texture_cube_map_array
GL_ARB_texture_query_lod
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ATI_texture_mirror_once
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_mirror_clamp
GL_EXT_texture_sRGB_decode
GL_NV_texture_barrier
</details>
--------------------------------------------------------

works well:
---------------------------------------------------------
PrusaSlicer
Version:   2.7.4+MacOS-x64
Build:     PrusaSlicer-2.7.4+MacOS-x64-202404050934

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Mojave Version 10.14.6 (Build 18G103)
Total RAM size [MB]: 8,590MB
OpenGL installation
GL version:   3.3.0 (3.3 NVIDIA-10.4.14 310.90.30.05b27)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GT 330M OpenGL Engine
GLSL version: 3.30.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_APPLE_client_storage
GL_APPLE_container_object_shareable
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_texture_range
GL_ARB_ES2_compatibility
GL_ARB_blend_func_extended
GL_ARB_draw_buffers_blend
GL_ARB_explicit_attrib_location
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_occlusion_query2
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shading_language_include
GL_ARB_texture_cube_map_array
GL_ARB_texture_query_lod
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ATI_texture_mirror_once
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_mirror_clamp
GL_EXT_texture_sRGB_decode
GL_NV_texture_barrier
</details>
-----------------------------------------------------------------

MagnusAbril, 2024-07-11:
Same issue

PrusaSlicer
Version:   2.8.0+win64
Build:     PrusaSlicer-2.8.0+win64-202406270929

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19045), 64-bit edition
Total RAM size [MB]: 17,133MB
OpenGL installation
GL version:   3.3.0 (3.3.0)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     GeForce 210/PCIe/SSE2
GLSL version: 3.30.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_clear_buffer_object
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
[... 173 lines cut ...]
GL_NV_texture_expand_normal
GL_NV_texture_multisample
GL_NV_texture_rectangle
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

lukasmatena, 2024-07-17:
We most likely found and fixed the problem. You are all welcome to test this build:

[PrusaSlicer-2.8.0+1-lm-spe2411-linux-x64-GTK3-g86e8e25-202407171335.AppImage](https://slicerbuilds.prusa3d.com/lm/PrusaSlicer-2.8.0+1-lm-spe2411-linux-x64-GTK3-g86e8e25-202407171335.AppImage)
[PrusaSlicer-2.8.0+1-lm-spe2411-MacOS-universal-g86e8e256a-202407171406.dmg](https://slicerbuilds.prusa3d.com/lm/PrusaSlicer-2.8.0+1-lm-spe2411-MacOS-universal-g86e8e256a-202407171406.dmg)
[PrusaSlicer-2.8.0+1-lm-spe2411-win64-g86e8e256a-202407171334.zip](https://slicerbuilds.prusa3d.com/lm/PrusaSlicer-2.8.0+1-lm-spe2411-win64-g86e8e256a-202407171334.zip)

(The Linux AppImage will not run on the latest Linux distributions which ship with libwebkit4.1. That is a separate issue.)

Thanks.

JavierBlancoGlez, 2024-07-17:
The 64-bit Windows version seems to be working fine, at least regarding the original issue. I haven't tested it thoroughly.

brigthub, 2024-07-17:
PrusaSlicer 2.8.0 spe2411 is now displaying a gcode preview on my windows10 setup.  It feels slower than version 2.7.1 and the layer slider control is super hard to use with the updates lagging behind.

dwprod38, 2024-07-17:
Hello, 

Same Error Message after launching: 
[image]

After clicking OK and loading 3mf file: can see slicing preview but very low fps refreshing preview : see two videos comparison,

PrusaSlicer 2.7.4
https://github.com/user-attachments/assets/c0efa9e9-a2da-43e9-8559-ea0fac68a798

PrusaSlicer 2.8.0
https://github.com/user-attachments/assets/6c6520ce-e6f9-4da2-8b00-ebbf7139485f

cheers
Stef

PrusaSlicer
Version:   2.8.0+1-lm-spe2411-MacOS-x64
Build:     PrusaSlicer-2.8.0+1-lm-spe2411-MacOS-x64-g86e8e256a-202407171406

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Sierra Version 10.12.6 (Build 16G2136)
Total RAM size [MB]: 8,590MB
OpenGL installation
GL version:   3.3.0 (3.3 NVIDIA-10.2.37 310.90.10.05b54)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GT 330M OpenGL Engine
GLSL version: 3.30.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_APPLE_client_storage
GL_APPLE_container_object_shareable
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_texture_range
GL_ARB_ES2_compatibility
GL_ARB_blend_func_extended
GL_ARB_draw_buffers_blend
GL_ARB_explicit_attrib_location
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_occlusion_query2
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shading_language_include
GL_ARB_texture_cube_map_array
GL_ARB_texture_query_lod
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ATI_texture_mirror_once
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_mirror_clamp
GL_EXT_texture_sRGB_decode
GL_NV_texture_barrier
</details>

fitimollas, 2024-07-18:
> We most likely found and fixed the problem. You are all welcome to test this build: [...]

It works very well in my case. W10. Thank you very much for the quick solution.

malowz, 2024-07-19:
> It works very well in my case. W10. Thank you very much for the quick solution.

Working fine now, thanks!

gmarconi2, 2024-07-20:
Now you can see the preview, it seems ok.

DerKameramann, 2024-07-22:
OSX Version on old Nvidia Graphics is working well now. No Performance issue any more, preview (inkl. layer slider control) works like a charm. Top Job, you found the issue for shure. Applause including standing ovations.
THX!

Golfsyn, 2024-07-30:
> We most likely found and fixed the problem. You are all welcome to test this build: [...]

It works fine, FPS is acceptable, thank you.
W10Pro 22H2
NVIDIA GeForce 9600 GT

pavolNovak, 2024-08-24:
Also had the same problem with older laptop, fixed now.

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

closed 2024-09-12 by lukasmatena

dwprod38, 2024-09-17:
For information to @lukasmatena : after installing  PrusaSlicer 2.8.1 rc1  --> G-code preview performance is very bad when moving bed with mouse (refreshing every 2 to 4 secs !!!) but in 3D editor view : everything move smoothly, slicing calculation speed unchanged comparing with 2.7.4. It seems openGL acceleration is missing or disabled: decision to downgrade to PrusaSlicer 2.7.4 

Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Sierra Version 10.12.6 (Build 16G2136)
Total RAM size [MB]: 8,590MB
OpenGL installation
GL version: 3.3.0 (3.3 NVIDIA-10.2.37 310.90.10.05b54)
Profile: Core
Vendor: NVIDIA Corporation
Renderer: NVIDIA GeForce GT 330M OpenGL Engine
GLSL version: 3.30.0
Textures compression: Enabled
Mutisampling: Enabled (4 samples)