prusa3d/PrusaSlicer #12911: Painted support blockers are not respected when model has been mirrored

Wombot, 2024-06-25:
### Description of the bug

When the object I am attempting to print is mirrored across the X or Y axes, via the PrusaSlicer right-click menu options, any painted support blockers are ignored and supports are generated. The issue does not persist if the object is then mirrored back to it's original orientation.

### Project file & How to reproduce

[files.zip](https://github.com/user-attachments/files/15979501/files.zip)

- Open project file
- Note support blockers painted on mounting holes on object underside
- Slice
- Note no supports are generated where support blockers were painted

- Return to 3D Editor View
- Select the object and right-click it
- Select Mirror > Along X Axis
- Inspect mirrored part, note support blockers are still painted around holes at bottom of object
- Re-slice the object (may need to move it around the bed to trigger the "Slice now" option)
- Note that supports have been generated in the holes where the support blockers were painted

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.4+win64

### Operating system

Windows 10 22H2 - 19045.4291

### Printer model

Prusa MINI+ (non-Input Shaper) 6.0.0

neophyl, 2024-06-25:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/12695

Jan-Soustruznik, 2024-06-28:
Hello, @Wombot, thank you for you report. Problem is already reported, closed issue as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/11980

closed 2024-06-28 by Jan-Soustruznik