prusa3d/PrusaSlicer #12912: PrusaSlicer 2.8.0-rc1 - Missing Borders around Menu Buttons

CantFindABatterman, 2024-06-26:
### Description of the bug

The Buttons on the top menu are missing borders. Without borders the menu is ambiguous and difficult to navigate. 

[image]

### Project file & How to reproduce

Download PrusaSlicer 2.8.0-rc1 unzip, open, install and updates

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.8.0-rc1

### Operating system

win 10

### Printer model

mk4 with IS

TerPuc, 2024-06-27:
Hello @CantFindABatterman, thank you for reporting. However, this is intended behavior of the new Modern UI. The active tab is highlighted. If you place the cursor over the other tabs, they will be highlighted with sligtly lighter shade for better navigation. From our point of view, this is not an error and we can close this issue. Thank you again for your time and have a nice day!

closed 2024-06-27 by TerPuc

CantFindABatterman, 2024-06-27:
That's an unfortunate decision. It gives the interface a haphazard and amateur appearance, you guys should change it back.

dncoder, 2024-06-27:
Agree. New UI looks awful :(  So much waste of space. Now to access menus, for example to open a recent file, I have to click "Menu" first, and only then move the cursor under "File". What was the point of this change? Why just not leave an option to use an old normal UI?