prusa3d/PrusaSlicer #12913: Last Top Layer Before Infill Is Mostly Missing

MechanicalCat, 2024-06-26:
### Description of the bug

Top fill is set to rectilinear. But the 7th layer (which is the top layer of a top surface) is partially concentric, matching some features which were present in the 1st layer. The 7th layer is also extremely sparse. A warning "Detected print stability issues: Long bridging extrusions" is popped up. This appears to be because the first layer of infill (8th layer) is mostly printing onto air, due to the absence of most of the 7th layer.
Expected behaviour: 7th layer would be rectilinear and covering 100% of the surface.

### Project file & How to reproduce

[Desktop.zip](https://github.com/user-attachments/files/15990944/Desktop.zip)

3mf and png are in the zip.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.3+linux-x64-GTK3

### Operating system

Kubuntu 22.04 6.5.0-28-Generic (64-bit)

### Printer model

Prusa i3 Mk3(?)

u89djt, 2024-06-26:
(fellow user) It's actually warning about layer 4 - you can see that those bridge extrusions in blue are very long continuous extrusions. But, it's fine, because they've got plenty of support points beneatht hem. I guess the point of a warning is that you ought to look closely and work out for yourself whether you can go ahead sensibly.
[image]
Layer 7 is indeed a bit more exciting than you might hope, but the next layer of infill looks like it should succeed just fine. If you move the bottom slider you can watch it draw bold long strokes that should bridge those gaps without difficulty unless your filament is wet/poor tolerance.
[image]

Snuff1eupagus, 2024-06-27:
u89djt wrote:
Layer 7 looks a bit more exciting than you might hope,

I found this comment very funny, and I do agree. Quite a strange way this was handled. its acts the same in 2.8.0 rc2

u89djt, 2024-06-27:
Heh :) Yeah, I guess it's like so many systems that have to deal with real unforeseen people/objects: why on earth is it doing that? Because it will routinely be impossible to do it the seemingly easy way, or that state of knowledge you are basing your impression on doesn't appear in the code or data explicitly.

Jan-Soustruznik, 2024-06-27:
Hello, @MechanicalCat , thank you for you report. Similar issue has been already reported in https://github.com/prusa3d/PrusaSlicer/issues/11629  I include it to main topic for collection models. Thank you, have a nice day

closed 2024-06-27 by Jan-Soustruznik