prusa3d/PrusaSlicer #12915: Adding SVG Modifier broken on 2.8.0-rc1 Mac

NLTMW, 2024-06-26:
### Description of the bug

2.8.0-rc1 on Mac 
Right click to add SVG modifier to a simple object.
The imported SVG is not attached to the main object
Option to "Use Surface" is ghosted
[image]
[image]

### Project file & How to reproduce

[SVG-error.3mf.zip](https://github.com/user-attachments/files/15994070/SVG-error.3mf.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0-rc1+MacOS-arm64

### Operating system

masOS Sonoma 14.5

### Printer model

PrusaXL 5TH

Snuff1eupagus, 2024-06-27:
Can you post the .svg ? unfortunately since it's already been imported it's not possible to see the original size or even import it again to test if it's working on other OS's.

NLTMW, 2024-06-27:
[STL_and_SVG.zip](https://github.com/user-attachments/files/16014234/STL_and_SVG.zip)
Sorry, I am a GitHub rookie. I hope this is the right place to send the STL and SVG.

Snuff1eupagus, 2024-06-28:
> [STL_and_SVG.zip](https://github.com/user-attachments/files/16014234/STL_and_SVG.zip) Sorry, I am a GitHub rookie. I hope this is the right place to send the STL and SVG.

No worries, I'm a user here not affiliated with Prusa, I sometimes try to solve issues or want to see how deep the rabbit hole goes, I'm experiencing a great deal of issues with the 2.8.X releases and frankly am very unhappy that they just released the 2.8.0 final with almost all the major issues still present. It appears Prusa's dev team is trying to win a race and doesn't care how they get there. It's a very poor release.

As to this Issue, I am experiencing the same behavior you reported on every version of Windows as well 7/8.1/10 and 11.
Thank you for up loading the .svg I wanted to load it several different ways, I tried as Neg Vol, Part and Modifier.

NLTMW, 2024-06-28:
Thank you for reassuring me that I am not doing something wrong, and there is a problem with that feature. Hopefully they release a better version soon.

barzto, 2024-06-28:
There seams to be something broken with the SVG modifier being invoked from 3D view. To workaround it for now, you can try to invoke the SVG modifier from outline in right side panel instead, by right click on the object you want to place SVG onto.

NLTMW, 2024-06-29:
Using the modifier from the right pane worked! Thanks for sharing that work-around!

Snuff1eupagus, 2024-06-29:
@barzto
not that I don't appreciate your fix. But this 
should·n't be an issue.

Snuff1eupagus, 2024-06-29:
You have shown nothing that shows that you have control .Over the software they release. good luck you poor simpleton users, and you are that, you have bowed to your superiors, and have no idea how to use this software you have been provided, you poor basters. I feel for you, and you are lost.

Snuff1eupagus, 2024-06-29:
I resign form these forums, its no wander PRUSA proceeds the way they do.

kubispe1, 2024-07-31:
This issue will be most probably fix with next release. SPE-2386
Thanks

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

closed 2024-09-12 by lukasmatena