prusa3d/PrusaSlicer #12916: PrusaSlicer 2.8.0-rc2 Individual Object & Part settings are not being respected

Snuff1eupagus, 2024-06-26:
### Description of the bug

When applying Individual Object setting "Seam position:" only the overall setting applied in Print Settings>Layers and perimeters>[Advanced]>Seam position: is taken into account and applied.

As well this same thing is happening with both the "Object" and "Part" settings for:
Only one perimeter on first layer:
and
Single perimeter on top surfaces:
only the overall settings applied in Print Settings>Layers and perimeters>[Only one perimeter] are respected

This has been a problem since the first public alpha, PrusaSlicer 2.8.0-alpha5
I have not checked all the Object and part settings for this problem but I'm willing to bet there are more.

BTW to whom it may concern, the Seams are finally looking good and are controllable again, TY

### Project file & How to reproduce

z

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.8.0-rc2

### Operating system

Windows 7/8.1/10/11 64bit

### Printer model

N/A

Jan-Soustruznik, 2024-06-27:
Hello, @Snuff1eupagus , thank you for you report. We will try to fix in some of next releases. Thank you 

Issue created: SPE-2384

jaguilar, 2024-08-03:
FWIW this also applies to seam position. I was trying to print some cylinders where a uniform seam leads to bad behavior and setting random seams on those objects did not work.

closed 2024-09-12 by lukasmatena