prusa3d/PrusaSlicer #12917: PrusaSlicer 2.8.0 rc2 crashes on upon closing win7 64

Snuff1eupagus, 2024-06-26:
### Description of the bug

The crash upon "opening" in Win 7 64 was fixed from the alpha5 version, however rc2 now crashes every time Upon Closing Win 7 64 

### Project file & How to reproduce

z

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.8.0-rc2

### Operating system

Windows 7 64

### Printer model

N/A

sdmacleod, 2024-07-16:
Just want to add that I'm experiencing this also, version 2.8.0

EricHedden, 2024-07-19:
I'm experiencing the same issue. 
Version 2.8.0+win64

BeastOldone, 2024-08-24:
Same here, crashes on close running Win7 64.

siskap1, 2024-11-16:
Same here, version 2.8.1, Win7 64bit

sdmacleod, 2024-12-26:
Seems to be fixed in 2.9.0

closed 2026-09-29 by prusaslicer-bot[bot]