prusa3d/PrusaSlicer #12918: Save colours of MMU - filament remember, color not

Flaknutej, 2024-06-27:
**Is your feature request related to a problem? Please describe.**
i am always frustrated, i need to change colors of mmu settings, i am not always near to printer, and i dont remember 1-white,2-blue.. etc.

**Describe the solution you'd like**
PS can remember type/profile of material, why not set right color?

**Describe how it would work**
Just set color, and slicer will remember the color you set

**Describe alternatives you've considered**
PS will ask you - do you want to save colors? yes/no... simple

peterjhl, 2024-06-29:
I second that

Hovedskou, 2024-06-29:
I also would like this feature for my XL5T very much. 
It is annoing to set the colors every time Prusaslicer is restarted.

closed 2026-09-29 by prusaslicer-bot[bot]