prusa3d/PrusaSlicer #12920: Prusa Slicer 2.8 rc 2 / iCCP: known incorrect sRGB profile

Martin-Ox, 2024-06-27:
### Description of the bug

When I start the new rc2 of Prusa Slicer 2.8, I get an error 
iCCP: known incorrect sRGB profile , and then the sysdate & time
Because I have 2 screens there are two lines of the error.

### Project file & How to reproduce

Just starting the prusa-slicer 2.8 r2 executable.

Videocard: Nvidia Geforce GTX 1660 Super
2x HP EliteDisplay E243

Drivers and firmwares are up to date.
[log.txt](https://github.com/user-attachments/files/16014931/log.txt)
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0-rc2+win64

### Operating system

Windows 10 22H2

### Printer model

Prusa MK4

mrbpoe, 2024-07-01:
Also experiencing the same issue with a GTX 3080 TI

RP2090, 2024-07-02:
Here the same problem. Maybe the custom color filament by default could be the cause? (RTX 2080Ti)

darthgloda, 2024-08-12:
Same issue on macOS 14.5 Build 23F79, 2019 Retina 5K iMac. PrusaSlicer 2.8.0+MacOS-x64.

ententeak, 2024-08-12:
Same with tripple screen setup - NVidia GeForce GT710 and GTX1660 - both at driver v. 27.21.14.5671 (latest for GT710)
Windows 11, build 22631.3958

darthgloda, 2024-09-16:
The error persists for me in 2.8.1RC1. Pops up as soon as I launch the slicer and then at regular intervals.

YuSanka, 2024-09-17:
SPE-2485

closed 2024-09-19 by lukasmatena