prusa3d/PrusaSlicer #12922: New PrusaSlicer 2.8.0 does not start in Linux

erickwill, 2024-06-27:
### Description of the bug

Starting the new version 2.8.0 does not open the app. Here is the message that I got:

/tmp/.mount_PrusaSL5R6gt/usr/bin/bin/prusa-slicer: error while loading shared libraries: libsoup-2.4.so.1: cannot open shared object file: No such file or directory

Note that previous versions works flawlessly.

### Project file & How to reproduce

It even can open the app.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Arch Linux

### Printer model

tronxy x5sa

erickwill, 2024-06-27:
For those using linux install both packages:
- webkit2gtk
- libsoup

closed 2024-06-27 by erickwill

smogol66, 2024-07-21:
Hello, 
I'm the same issue in ubuntu 24.04 LTS.

Normally, these library should be part of the **AppImage** and not relay on the installed OS libs.

Ubuntu uses the latest library libsoup3.0 so it is not possible to run this version.
` /tmp/.mount_PrusaSAC8t8X/usr/bin/bin/prusa-slicer: error while loading shared libraries: libsoup-2.4.so.1: cannot open shared object file: No such file or directory`

If I install the library libsoup2.4-common, I got following error:
`libsoup-ERROR **: 22:12:25.668: libsoup2 symbols detected. Using libsoup2 and libsoup3 in the same process is not supported. Trace/breakpoint trap(core dumped)`

Consider to reopen this issue as "not solved"

jovaska1337, 2024-11-20:
A simple fix would be to remove webkit2gtk-4.0 from `src/slic3r/CMakeLists.txt` (or to at least change the ordering to prefer 4.1) which is the cause of libsoup-2 being linked in. I'm not sure why it's there to begin with considering the minimum wxWidgets version (3.2) pulls in webkit2gtk-4.1 which depends on libsoup-3. I managed to build a working binary on Gentoo with this modification.