prusa3d/PrusaSlicer #12923: Sliced Layers Not Visible in Preview on PrusaSlicer 2.8.0 final release

JavierBlancoGlez, 2024-06-27:
### Description of the bug

I have recently installed PrusaSlicer version 2.8.0  and encountered an issue that I did not experience in previous versions. When I proceed to slice a model, the platform view, where the sliced layers should be displayed, appears empty. This issue occurs regardless of the model format, as I have tested with both STL and STP files. The slicing process completes without errors, but the preview of the layers does not show up. This bug significantly impacts the ability to verify and adjust the slicing settings before printing. Please investigate and resolve this issue.

### Project file & How to reproduce

[Test1.zip](https://github.com/user-attachments/files/16017738/Test1.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 10  X64

### Printer model

MK4

MrDixioner, 2024-06-27:
I downloaded your project and didn't find any problems. I also loaded my models in STL and STEP formats, and everything was fine with cutting and displaying layers.
I would like to see a screenshot of your problem, because it's hard to understand what's going on.

Megastrong1, 2024-06-27:
I have the same problem
[image]

N4ndo47, 2024-06-28:
Same thing happening to me

TerPuc, 2024-06-28:
Dear @N4ndo47, this issue is a duplicate of #12908. I am closing your issue but you can check the first one for updates. Thank you!

closed 2024-06-28 by TerPuc

uysal23, 2024-06-29:
exactly same thing happenning to me. prusa slicer 2.8.0 with win10 pro version 22H2

Maiquess, 2024-08-18:
Mesma coisa comigo.