prusa3d/PrusaSlicer #12924: Missing Arm64 build for PrusaSlicer 2.8.0

ErwanAliasr1, 2024-06-27:
### Description of the bug

The 2.8.0 version is lacking of the Arm64 linux build.
It was present in previous releases but disappear on this one.
Can you add it again please ?
Thx !

### Project file & How to reproduce

2.8.0 release is not featuring the Arm64 Linux build in its assets

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Linux

### Printer model

MK4

DiemDM, 2024-06-29:
Previous ARM versions on Raspberry Pi, including 2.7.4 which itself only runs sketchily, prompt for auto update but then download incompatible Linux X64 build - I presume this is an error in the update message descriptor.
Please provide an ARM build for the download target.

ErwanAliasr1, 2024-07-01:
Noticed that also. The CPU arch should be considered while considering a
new update.

ErwanAliasr1, 2024-10-15:
2.8.1 is not having ARM64 build

DiemDM, 2026-09-22:
Still no Arm build for 3.0 alpha - can we expect something when we get to betas?