prusa3d/PrusaSlicer #12925: Print Settings > Post-processiong Scripts: Can't access box to edit/add scripts

DangerManiac, 2024-06-27:
### Description of the bug

Cannot access edit box for Post-processing scripts. I currently have one in there that does run as it should but when looking at the edit box I don't see it listed nor can I edit it or add another script.

Fixed by manually editing the print settings .ini file and adding it in, this however is NOT ideal.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 10 x64

### Printer model

Ender 3 Pro, Ender 3 Max, Neptune 4

3DPrintMunich, 2024-06-27:
Same here

### Description of the bug

after editing the pp script the input field disappears.

### Project file & How to reproduce

[image]
[xyzCalibration_cube_output options.zip](https://github.com/user-attachments/files/16020140/xyzCalibration_cube_output.options.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Win 10

### Printer model

Snapmaker J1

Lodesys, 2024-06-28:
Seeing the same problem. Upgraded from 2.7.4 The post-processing script text is there, but just not displayed (like white text on white background). I can click in the field and copy or alter the text, but can't see what's happening. Running Windows 11 on an ASUS AMD G14 gaming laptop.

As a temporary work around... Click in the text box, select all text and copy it (Ctrl-A, Ctrl-C) into your text editor / Notepad. View the text and, if needed, make any changes & paste it back into PrusaSlicer (Ctrl-A, Ctrl-V).

ipcn, 2024-07-12:
I've faced the same issue with version 2.8.0 on Windows 10. It happens only if text in this filed is long enough, while short text does not disappear. It looks like height of this field is calculated incorrectly depending on the text length. If text is long enough, filed height is too big and text is not visible.

tol2cj, 2024-07-22:
I have similar issue, PS 2.8.0 on Win10 x64. I alreary have script in the profile from past for sending gcode to Sindoh 3DWOX 1 printer. I have to change IP, which is a parameter of the script. With the Copy - Paste method is possible but not comfortable.

[image]

this should be seen: "SindohPostScript.exe xxx.xxx.xxx.xxx;"

lukasmatena, 2024-08-28:
Closing as a duplicate of #12932. It will be fixed in 2.8.1, which will be released soon.

closed 2024-08-28 by lukasmatena