prusa3d/PrusaSlicer #12926: "Physical Printer" tab is loading forever in PrusaSlicer 2.8.0

DominikPalo, 2024-06-27:
### Description of the bug

I updated my PrusaSlicer from version 2.7.3 to 2.8.0, opened it and after clicking on the "Physical Printer" tab it shows a spinning spool animation without ever loading the content.

[image]

### Project file & How to reproduce

N/A

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+MacOS-arm64

### Operating system

macOS 14.3.1 (23D60)

### Printer model

MK4

mix579, 2024-06-28:
I've observed the same. Somehow managed to get it to work but don't know how.

fjpanag, 2024-06-29:
I am experiencing the same.

I am trying to connect to Mainsail for a Klipper-based Voron.

Sometimes I see the Mainsail page loading briefly. But it always ends up to a blank page:

[image]

Here is my system details:

```
PrusaSlicer
Version:   2.8.0+linux-x64-GTK3
Build:     PrusaSlicer-2.8.0+linux-x64-GTK3-202406271021

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 6.9.5-200.fc40.x86_64 x86_64
Total RAM size [MB]: 33,326MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 550.90.07)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1070/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
[... 325 lines cut ...]
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vdpau_interop
GL_NV_vdpau_interop2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
</details>
```

Mendes11, 2024-09-28:
I just installed Prusa Slicer. I was able to connect to my Repetier Server through the server configuration, the test returned an OK, but once I clicked on "Physical Printer", nothing showed up, just a spinning wheel when you hover over the tab...
I'm using V2.8.1.

```
PrusaSlicer
Version:   2.8.1+linux-x64-GTK3
Build:     PrusaSlicer-2.8.1+linux-x64-GTK3-202409181416

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 6.8.0-45-generic x86_64
Total RAM size [MB]: 33,548MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 550.107.02)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1650/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
[... 327 lines cut ...]
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vdpau_interop
GL_NV_vdpau_interop2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
</details>
```

closed 2026-09-29 by prusaslicer-bot[bot]