prusa3d/PrusaSlicer #12927: Add the ability to refresh a model from file

techis, 2024-06-27:
**Is your feature request related to a problem? Please describe.**
I am frustrated when I am trying to rapidly prototype a new design and have to remove and re-add the model. Then reapply any slicer customizations I made (orientation, cuts, supports, etc.).

**Describe the solution you'd like**
I would like the ability to refresh the model from the file. This has a base requirement that the file is named the same between iterations.

**Describe how it would work**
1) a model is created and added to the build plate
2) it is sliced and printed. Changes are required and the model is updated.
3) new model file replaces the original file that was added to Prusa slicer. A button is added to the Prusa Slicer interface for enabling Rapid Prototyping that when checked, allows the user to refresh the models from file. 
4) Click the refresh model button. New file is loaded with any changes attached. 
5) repeat steps 2-4 as needed.

**Describe alternatives you've considered**
I have considered using the command line slicing utility however this seems overly cumbersome and involved. Removing and re-adding the model is easier.

techis, 2024-06-27:
Thanks guys. I often find that when initially designing something, I need to print multiple iterations to find the "just right" measurements.

Jan-Soustruznik, 2024-06-28:
Issue closed, problem solved.

closed 2024-06-28 by Jan-Soustruznik