prusa3d/PrusaSlicer #12929: file transfer abort

crashkg, 2024-06-27:
### Description of the bug

When transferring a file from prusaslicer I get a file transfer abort error

### Project file & How to reproduce

This has happened on mulitple files

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.80

### Operating system

Windows 11

### Printer model

Prusa XL, and MK4

Jan-Soustruznik, 2024-06-28:
Hello, @crashkg , could you share some more info about the issue. thank you

crashkg, 2024-06-28:
It is intermttent. I'm trying to decipher why it's happening. When it crops up again I will update. It is not an issue with my internet as that seems to be working fine. The files end up on the printer even though I get the error, and they print fine. I am on the latest firmware and prusaslicer.

closed 2026-09-29 by prusaslicer-bot[bot]