prusa3d/PrusaSlicer #12930: Prusa Slicer 2.80 - Seam Positions (REAR)

DJ-Miller, 2024-06-27:
### Description of the bug

Seam positions set to REAR on two critical part surfaces.  Seams are no longer at the correct position after installation of Prusa Slicer 2.80 this morning.

### Project file & How to reproduce

[2.80 Prusa Slicer Update - Seam Positions.pdf](https://github.com/user-attachments/files/16019202/2.80.Prusa.Slicer.Update.-.Seam.Positions.pdf)
[33_Parts_Color_Logo_BUG_2,80.zip](https://github.com/user-attachments/files/16019256/33_Parts_Color_Logo_BUG_2.80.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Windows 11 Pro

### Printer model

Prusa MK4 with MMU3

DJ-Miller, 2024-06-27:
The seam position on the two indicated parts are planned to be at a specific rotational position when the model is manipulated in order to give a visual indication of rotation.  This model has been printed many times before with great results, but in opening the project in the newly installed 2.80 software update, the seams no longer show as aligned with the REAR on these two parts.

Snuff1eupagus, 2024-06-27:
This is s a duplicate of  https://github.com/prusa3d/PrusaSlicer/issues/12916

To add to the Issue I posted the other day I noticed that if I imported a file prom an older properly working version of slicer sometimes the seams would act as expected, until I moved or adjusted something in the file.

Jan-Soustruznik, 2024-06-28:
Hello, @DJ-Miller , thank you for for report. We know about the issue. We will try to fix in some of next release. Have a nice day 

duplicte of https://github.com/prusa3d/PrusaSlicer/issues/12916

closed 2024-06-28 by Jan-Soustruznik