prusa3d/PrusaSlicer #12931: Appimage error PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929.AppImage

juckendesAuge, 2024-06-27:
### Description of the bug

2.8.0 have under Fedora 40 an error. 2.7.4 works fine.

md5sum PrusaSlicer-2.7.4+linux-x64-GTK3-202404050928.AppImage 
876343150dc2c3e7590363988932b63b  PrusaSlicer-2.7.4+linux-x64-GTK3-202404050928.AppImage

md5sum PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929.AppImage 
2a0bce7fcd05edcafc9801af0db79470  PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929.AppImage

./PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929.AppImage 
/tmp/.mount_PrusaSxYOK4W/usr/bin/bin/prusa-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

### Project file & How to reproduce

./PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929.AppImage 
/tmp/.mount_PrusaSxYOK4W/usr/bin/bin/prusa-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929.AppImage

### Operating system

Fedora release 40 (Forty)

### Printer model

Prusa MK3s+

richarrd-h, 2024-06-27:
I have the same problem with slackware 15.0

jfbauer432, 2024-06-27:
Same issue on Ubuntu 24.04

alsternerd, 2024-06-28:
Same on Arch Linux.

lukasmatena, 2024-06-28:
There are two AppImages in the [release page](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0), you need to download the other one. I have just amended the release notes with a paragraph describing the situation.

juckendesAuge, 2024-06-28:
Thank you very much 

But the name Ubuntu is a bad choice if it is to be used on several distributions. For this reason, I have not even tried to test the version...
It only refers to one Linux (Ubuntu) distribution.

alsternerd, 2024-06-28:
> There are two AppImages in the [release page](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0), you need to download the other one. I have just amended the release notes with a paragraph describing the situation.

It's the same with the version stated in the title, not just the Ubuntu-version.

On what system did you test the .AppImages before releasing?
Since libwebkit2gtk isn't even a package available in pacman on Arch. Haven't tested on Debian (stable and testing), yet.

lukasmatena, 2024-06-28:
Thanks for your comments. I understand that the situation is not ideal and that there is room for improvement.

> But the name Ubuntu is a bad choice if it is to be used on several distributions.

I understand. However, listing all distributions in the name would not be better, not to mention impossible. A suffix like "Ubuntu-24-04_Fedora-40_Slackware-15_..." did not sound right either. The "Ubuntu-24-04" was supposed to mean that the binary was **built** on Ubuntu 24.04. We briefly considered some "old" / "new", but that would probably raise even more questions as to what new features there are, etc.

We will spend some more time trying to build the AppImage that just runs everywhere. If we do not succeed, we may reconsider if AppImage is a good format to use for PrusaSlicer deployment... And if not, what is.

alsternerd, 2024-06-28:
> We will spend some more time trying to build the AppImage that just runs everywhere. If we do not succeed, we may reconsider if AppImage is a good format to use for PrusaSlicer deployment... And if not, what is.

The .AppImage format is okay and better than snap or something like that. 
But what can be done immediately is state the dependencies, in this case >= libwebkit2gtk-4.0 could be enough.
Maybe even in the Updatedownloader inside the PrusaSlicer.AppImage, so that we don't run into that after the update. :)

Drezir, 2024-06-28:
What about flatpak? Is it an option to pack prusaslicer?

sam-e, 2024-06-28:
Still not working on Fedora 40

neclimdul, 2024-06-28:
I don't know if its new but there's a separate appimage for these distros to fix this. It worked great for me. 

PrusaSlicer-2.8.0+linux-x64-GTK3-202406271021-Ubuntu-24-04.AppImage

BTW, Thanks Prusa! Orca's been broken by this for quite a while now but you've already worked around it.

skint007, 2024-06-29:
I get this error when running the one for Ubuntu 24.04, which I'm on, if I run the none Ubuntu one, it just closes out instantly after running.
[image]

nekopep, 2024-06-29:
Under debian 12 we have:
/usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0
instead of 4.0.so

sam-e, 2024-06-29:
The ubuntu version works if you create your own .desktop file, the appimage is not compatible with "appimagelauncher"

abatkin, 2024-06-30:
> The ubuntu version works if you create your own .desktop file, the appimage is not compatible with "appimagelauncher"

You don't even need to create your own .desktop file - launch it once (i.e. via a terminal) and let it do its own desktop integration.

trainmeditations, 2024-06-30:
> There are two AppImages in the [release page](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0), you need to download the other one. I have just amended the release notes with a paragraph describing the situation.

Neither appimage works in Debian 12 (bookworm). 

PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929.AppImage  doesn't work with the system libwebkit2gtk
PrusaSlicer-2.8.0+linux-x64-GTK3-202406271021-Ubuntu-24-04.AppImage doesn't work with the system glibc

```
/tmp/.mount_PrusaSLcDndp/usr/bin/bin/prusa-slicer: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by /tmp/.mount_PrusaSLcDndp/usr/bin/bin/prusa-slicer)
/tmp/.mount_PrusaSLcDndp/usr/bin/bin/prusa-slicer: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_PrusaSLcDndp/usr/bin/bin/prusa-slicer)
/tmp/.mount_PrusaSLcDndp/usr/bin/bin/prusa-slicer: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_PrusaSLcDndp/usr/bin/bin/prusa-slicer)
```

I tried using flatpak, but the entrance script for that isn't reading from the xdg doc mount. The files are placed there correctly and if I launch it on its own I can open a file from there, but it's not launching properly when launched from a file.

Doesn't seem to be a working installation option for Debian at the moment

juckendesAuge, 2024-06-30:
> Still not working on Fedora 40

Flatpak works fine under Fedora 

Flatpak—the future of application distribution - https://www.flatpak.org/setup/Fedora

nekopep, 2024-06-30:
I think on debian 12 asking for libwebkit2gtk-4.1 will do the job

sam-e, 2024-06-30:
> > Still not working on Fedora 40
> 
> Flatpak works fine under Fedora
> 
> Flatpak—the future of application distribution - https://www.flatpak.org/setup/Fedora

v2.8 is not available with flatpack

flux242, 2024-07-11:
Both versions of the Appimage (Ubuntu and sansUbuntu) won't start on Ubuntu 22.04. Same error - missing libwebkit2gtk-4.1-0. I do not have that lib on my system at all!

First of all creating different versions of AppImage defeats its purpose! 
Secondly I don't really understand what's the problem of putting that lib into the container but instead starting to look for another container format! It worked before and now not? What is going on?

frank-pian, 2024-07-18:
Orca tried to package libwebkit in the appimage, and the size of the appimage increased by 50MB, which seems to be acceptable.
https://github.com/buzzhuzz/OrcaSlicer/actions/runs/8849590037
But they didn't, choosing to do the same as prusa.
https://github.com/SoftFever/OrcaSlicer/pull/5203

wohali, 2024-07-22:
> Doesn't seem to be a working installation option for Debian at the moment

Confirmed on Debian bookworm (12.6) here - neither version works, with the same errors as @trainmeditations posted. This is definitely a packaging issue.

Following [the build from source instructions](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md), I'm able to build and run 2.8.0 on the same machine.

umk0m1qk, 2024-08-02:
> > The ubuntu version works if you create your own .desktop file, the appimage is not compatible with "appimagelauncher"
> 
> You don't even need to create your own .desktop file - launch it once (i.e. via a terminal) and let it do its own desktop integration.

I have the exact same problem. Running 24.04 with appimagelauncher handling my appimage integrations. Can you give the specific CLI command to launch Prusa Slicer appimage bypassing appiamgelauncher? Uninstalling  appiamgelauncher isn't really an option since it will break all my other appimage integrations. As of right now, I can't get Prusa Slicer to work at all on Ubuntu 24.04. Yes, I am using PrusaSlicer-2.8.0+linux-x64-GTK3-202406271021-Ubuntu-24-04.AppImage

flux242, 2024-08-03:
Why do you need webkit? To render what's new page or to show ads? Can there be a version without that useful feature like before?

neclimdul, 2024-08-03:
@flux242 it's used for the printer view where you can see klipper ui's and prusa's management site.

VasilNikolov, 2024-08-12:
Can confirm debian doens't work even after installing libwebkit2gtk-4.1

probonopd, 2024-12-08:
To everyone affected, please try the AppImage linked in

* https://github.com/prusa3d/PrusaSlicer/issues/13653

and please let us know how how it works for you. Thanks!

nekopep, 2024-12-09:
I confirm it works on debian 12! Good job!

wohali, 2026-09-18:
It would be nice if PrusaSlicer continued Appimage builds. This is a comment to make it clear that there is still community interest in this, though there are community builds available for now -- we have no way of knowing if these will still be possible in the 3.0.0 codebase.

lukasmatena, 2026-09-23:
@wohali Hi. There is a separate discussion thread about AppImage: https://github.com/prusa3d/PrusaSlicer/discussions/15714. The community builds are still possible - source code of the application is still completely public. Closing this issue.

closed 2026-09-23 by lukasmatena