prusa3d/PrusaSlicer #12932: Output options disappear after editing

3DPrintMunich, 2024-06-27:
### Description of the bug

When editing the output options. The inpu field disappears.

### Project file & How to reproduce

[image]
[xyzCalibration_cube_output options.zip](https://github.com/user-attachments/files/16020140/xyzCalibration_cube_output.options.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Win 10

### Printer model

Snapmaker J1

ipcn, 2024-07-12:
I've faced the same issue with version 2.8.0 on Windows 10. It happens only if text in this filed is long enough, while short text does not disappear. It looks like height of this field is calculated incorrectly depending on the text length. If text is long enough, filed height is too big and text is not visible.

A workaround is to click on this field, press Ctrl+A and Delete, switch to another tab and then return back to this tab, and you'll see the original empty field.

YuSanka, 2024-07-17:
Thanks for your report. We can reproduce this issue.
SPE-2416

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

closed 2024-09-12 by lukasmatena