prusa3d/PrusaSlicer #12934: After slicing, the piece becomes invisible on the table.

OneTouch3D, 2024-06-28:
### Description of the bug

Após atualização para Prusa Versão 2.8.0+win64 o sistema ao fatiar a peça ela não aparece na mesa fatiada, os menus estão todos ok apenas a peça some.
Windows 10 Pro 

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Versão 2.8.0+win64

### Operating system

Windows 10 Pro 64

### Printer model

Ender 3 MAX

malowz, 2024-06-28:
this happens to me to, in the beta, alpha and now the 2.8 version. windows 7 x64. Ender 3 v2 neo

tried to add new printer, use standard presets, nothing work

video showing the problem:
[https://storage.googleapis.com/prusa3d-content-prod-14e8-wordpress-forum-prod/2024/06/4034fdfb-lmc_20240622_211036137.mp4](url)

2.7 versions work fine.

TerPuc, 2024-06-28:
Dear @OneTouch3D and @OneTouch3D, this issue is a duplicate of #12908. I am closing your issue but you can check the first one for updates. Thank you!

closed 2024-06-28 by TerPuc