prusa3d/PrusaSlicer #12937: Poor performance on Preview window

gusdleon, 2024-06-28:
### Description of the bug

In the last version of PrusaSlicer I noticed a downgrade of the performance in the preview not present in previews stable versions

### Project file & How to reproduce

1. Right click to add a new shape, any will work lets work with cilinder
2. change to preview should work as espected you will be able the scroll with the arrow keys trough the layers and individual lines, combine with ctrl or shift and will go faster
3. go back to plater and scale the cilinder to let say 100mm in Y
4. change again to preview and scroll again with keyboard, in should be laggy and if you keep pressed the keys it will simply not render another frame until you release the key/s

video showing the behavior
https://github.com/prusa3d/PrusaSlicer/assets/3170940/c1c748d2-093b-4fe0-9724-1ba707c712d6

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 10 (more info in the unique installation id CDA399EE8D05AFD06831CD********** )

### Printer model

N/A (but mk3s)

fe60, 2024-09-03:
According to the comments in #13076 Windows 11, Linux and MacOS seem to be affected too.

fe60, 2024-09-03:
For those who don't want to download [2.7.4 from Github](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.4), it's still available on [prusa3d.com](https://www.prusa3d.com).
[https://cdn.prusa3d.com/downloads/drivers/prusa3d_win_2_7_4.exe](https://cdn.prusa3d.com/downloads/drivers/prusa3d_win_2_7_4.exe)

gusdleon, 2026-09-19:
still a issue