prusa3d/PrusaSlicer #12938: "Export plate as STL/OBJ including supports" doesn't work like before

Gero003, 2024-06-28:
### Description of the bug

The problem is that since versions 2.6 and later, if I open or create a new project with several objects with their respective supports. Previously, only the selected and visible model was exported with its respective supports, but now the objects that were hidden within the same file are also added.
If I use a version like 2.5.2, this problem does not appear.

Video Test
[https://youtu.be/B648WbA3W-Q](url)

### Project file & How to reproduce

-Open the Project file
-Select the visible object and press "Slicer"
-Press File, Export, Export plate as STL/OBJ including supports

Try using different versions of the program to see the difference.
[Test.zip](https://github.com/user-attachments/files/16023836/Test.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 / 2.7.0/ 2.8.0

### Operating system

Windows 11

### Printer model

Prusa MK3S, Halot Mage

Jan-Soustruznik, 2024-06-28:
Hello, @Gero003 , thank you for your report. We will try to fix in some of next release. Have a nice day 

issue created: SPE-2388

positron96, 2025-04-20:
Just want to check if there was any progress since Jun 2024. 
So far the workaround for me is to remove hidden objects, export plate with supports, and undo to bring back removed objects. If I forget to undo and save, my hidden objects are lost, which is rather sub-optimal.

positron96, 2026-09-18:
Well, I can't find this feature in 3.0-alpha anymore.