prusa3d/PrusaSlicer #12939: -3

revaip, 2024-06-28:
### Description of the bug

I cannot enter "3" or "-" characters from the alphanumeric keyboard.

### Project file & How to reproduce

I cannot enter "3" or "-" characters from the alphanumeric keyboard eg. in the Cut and Arrange windows, infill settings on the main screen and in the regular window layout in any tab and any input boxes. The 3 and - keys on numeric part of the keyboard works.

The problem doesn't affect the non-modal window, so the problem is somewhere in main window code.

The problem persists in Hungarian and English-US keyboard layouts, too (the "-" takes place in different position (in the Hungarian layout, it's next to the right shift), so I think it's not keyboard escape code problem. Shift+3 ("+") and Shift+- ("_") works, just the - and 3 chacarters that cannot be entered.

We could reproduce it on a different computer, too (mine: win10x64, that one: win11).

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

windows 10 22H2 b19045. 4529

### Printer model

MK3S and MK4, but it's really irrelevant (not affected by the printer profile)

LordOfInsanity, 2024-07-02:
I have the same issue, also with the "-" not working but instead of the "3" the "1" isn't working on two different computers.
Same with the shift-combinations working just fine and the NumPad versions of both keys functioning just fine.
Tried it with multiple different Keyboards and Keyboard-Layouts (German, US-English, Greek)

allready had the issue in 2.8.0-alpha5 and still have it in the 2.8.0 released version
Windows 11, 23H2 build 22631.3737

lukasmatena, 2024-07-10:
Closing in favor of #13043, which contains a list of all the duplicates.

closed 2024-07-10 by lukasmatena