prusa3d/PrusaSlicer #12941: Prusa Slicer 2.8.0 Ender - 3v2 neo does not see gcode files

PrzemSek, 2024-06-28:
### Description of the bug

After updating from 2.7 to 2.8.0 my Ender 3v2 neo does not see gcode files.
I do everything as before. After cutting, I generate gcode and save it to the SD card. The file is on the card but after inserting it into the printer, Ender does not see the file. 
Files from other slicers are read normally.
How to fix it?

### Project file & How to reproduce

[klucz do kół RC.zip](https://github.com/user-attachments/files/16028521/klucz.do.kol.RC.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 7 64bit

### Printer model

Ender 3v2 Neo Mriscoc soft

malowz, 2024-06-28:
can you post also de gcode file that the printer didn't find?

PrzemSek, 2024-06-28:
The printer did not see any gcode. I tried on various projects.
I uninstalled Prusa, installed it again and suddenly everything was back to normal. Weird.

closed 2024-06-28 by PrzemSek