prusa3d/PrusaSlicer #12942: Negative text object name does not get updated after editing

nmaggioni, 2024-06-28:
### Description of the bug

When adding or copying a negative volume text object and editing its content, the name of the entity in the parts list side panel is not updated until a change is tried to be made to it. The custom name is then immediately discarded and the new, automatically assigned name correctly reflects the new text contents.

https://github.com/prusa3d/PrusaSlicer/assets/3594528/50d8201d-5f54-47f3-b189-763a5972d95d

### Project file & How to reproduce

1. Add Shape > Box.
2. (On the newly created part) Add negative volume > Text.
3. Edit the text to contain a custom string and close the dialog.
4. Notice how the name of the entity in the sidebar is still "Text - Embossed text".
5. Click on the name of the entity in the sidebar and rename it to a different value (or just click away without editing the original value).
6. Notice how triggering the edit made the correct object name appear.

These steps are shown in the video above by copying and pasting the same text object inside an existing part.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

Manjaro Linux

### Printer model

Any

Jony01, 2024-07-11:
I confirm, it is reproducible.

nmaggioni, 2024-07-12:
Seems to persist in v2.8.0, and text in the form of positive shape misbehaves too: on edit the label `Text - ` is prepended each time a modification is made even if it's already present. Repeatedly double clicking on the object name and then clicking out without making edits results in something like this: `Text - Text - Text - Text - Embossed text`.

closed 2026-09-29 by prusaslicer-bot[bot]