prusa3d/PrusaSlicer #12943: Bring back the old UI

wikolii, 2024-06-28:
Bring back the old UI, or at least the option to have the menu items like before.

Shinomorisama, 2024-06-28:
Came here specifically to ask for this.  We import and export 3mf files for every print for archiving in our work flows. It is tremendously helpful when you're trying to remember exactly how you did variable layer height setting changes or custom supports or trying to diagnose why something failed in a weird way.   The new menu is going to be annoying for anyone that uses options in there regularly.   I've only run 2 prints so far and it is annoying me enough to come post about it.

apopticon, 2024-06-28:
Hiding the menu behind another button is NOT an improvement.  I don't know of anyone using Prusaslicer on a smartphone, so why use UI elements designed for one? PLEASE give us back the menu bar. The extra clicks required to use the menu are just an annoyance that adds NOTHING to the UI.

dncoder, 2024-06-28:
I absolutely agree to have an option to choose a previous UI, which is more straightforward and looks much better.
Now you always have to click Menu first to access every option.
These large labels without borders and icons are not intuitive for fast naviagation. Maybe they would make more sense for touch devices but not for PC users.

cozmo14047, 2024-06-28:
Came here to say this, the new ui is terrible

paulcobbaut, 2024-06-28:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/12838
(35 years of keyboard muscle memory down the drain)

lcemoore, 2024-07-02:
This new GUI is not an improvement and the layout options in the preferences do nothing to make it better.

devfoo-one, 2024-07-02:
👍 I was really confused that there is no way to hide this top bar. Takes far too much screen estate.

diclegend, 2024-07-02:
Having to click a button first before being able to access the menu is a huge step backwards.  Please don't degrade this program and create inefficiencies by taking away functionality- bring back the ability to access the menu via keyboard shortcuts

Novahex451, 2024-07-02:
Hello... Dev team. I created a GitHub account just so I could comment on update 2.8. Like, what the heck? I have been using the Alt key to open menus since the day Windows 95 was released, and you thought it was a good idea to remove this ubiquitous functionality?

Make the world right again by bringing back core keyboard functionality, Thank you.

agmcmll, 2024-07-03:
I also created an account on GitHub to chime in with my agreement that the UI changes are a regression.  Just because google chrome decided to ruin their browser UI and firefox followed suit, and now Prusaslicer?  A program for people who create things?  The flimsy justification for ruining the UI in  browser was that it was "cleaner" and browsers are for consuming not creating.  Prusaslicer doesn't even have this justification.  At least in firefox there remains an about:config setting where I can bring a proper menu back. At least give us that even if you must serve up a crippled UI by default.

larrybud2004, 2024-07-05:
Crazy that they abandoned all standards of UI design. Put it back the way it was.

MR2C280, 2024-07-06:
Add my vote to all the others - please restore the normal menu list.  That drop-down menu just adds unnecessary mouse movements.

Just-ForFun, 2024-07-06:
Add my vote as well.  I thought I had a failed install for a bit.

Please bring back the old UI.

elomech, 2024-07-06:
Hello Dev team. I also created a GitHub account to comment on update 2.8. This UI isn't cleaner. This UI makes more work than necessary. My first attempt to find the right menu was annoying. "ALT" key no longer works either. The operation of the PrusSlicer has slowed down with this change and has become more cumbersome.

Please give us back the old UI!

cozmo14047, 2024-07-06:
Also to note over on prusa3d redit heaps of people are not happy about the ui change

SangreDeTruchas, 2024-07-07:
I sincerely hope to see this sad mistake undone. I'd love insight into how such a decision can be made, I just don't believe anyone who is both a software developer and a user of PrusaSlicer would ever consider a change like this. Having been told now that this is just a small part of a UI overhaul is greatly concerning to me; if this is is a taste of what's to come, then I surely won't have the palate for it. It's clear I'm not alone in that given the discussions on Reddit, the Prusa Forums, and here.

neclimdul, 2024-07-08:
Try to remember, people worked really hard on this change. People that clearly want to make PrusaSlicer better. Harsh criticism can be really painful and drive people that would be improving your software away from making the changes you want. As a general issue queue rule, especially in open source where its not a big corporation, consider the developer on the other side and to quote Wil Wheaton, "Dont be a ...."

---
Actual feedback, while I'm not a fan of the new UI, I do appreciate the effort. I spent countless numbers of hours complaining about the old UI as I learned it and still find it confusing, only able to find them through the search. With that in mind, making an effort to improve it is great and I think rather then reverting and bringing back the old UI I'd say iterate and continue to make it better.

Pluses:
 - I think I will like the new connect features. Its currently not functional for me but I know that's be worked on.
 - I definitely appreciate the new Physical Printer tab. It also currently doesn't wok but I look forward to it working better. I always have a browser open to switch to and this will save some clicks and window juggling.
 - I also like the new mode setting. Much more standard ui component and makes more sense.
 - Search box is a big improvement and as noted I think the search is a key feature of PrusaSlicer. Its always been there and I'll probably continue to use the hotkey but highlighting this hidden feature for new users is a great idea.

Minus:
 - I really don't like the menu. If it was going to be "hidden" I'd much prefer it did something like Firefox or a lot of other programs where its not visible but pushing alt makes it visible as the normal "Menu Bar" that's existed for years. This also makes it very accessible to keyboard only users. Something that might be a concern for EAA or other accessibility law compliance.
 - I think the tab structure is just hidden and still confusing. I don't know that the pane option is actually better, it adds more clutter to an already busy UI. I find Orca's attempt as well as Cura to be  frustrating as things are often too small to interact with and it is difficult to focus on. So... don't know about this one.

jurassic73, 2024-07-09:
Another vote to go back... I'm sure this made sense to someone but not liking it. At least give us a check box to use the 'classic' menu bar.

naldarn, 2024-07-09:
This type of change is what has sunk many UI in the past, the new menu structure slows down use of the software and obfuscates funtions used on every project.

Please stop making changes to things that don't matter and look into other issues before making useless changes to the UI.

larrybud2004, 2024-07-09:
> This type of change is what has sunk many UI in the past, the new menu structure slows down use of the software and obfuscates funtions used on every project.

What's amazing is that the new menu is so out of the norm of common UI/menu design. NOBODY has a menu like this, completely unintuitive.

mrwizardtx, 2024-07-11:
The new Gui/menu is user hostile, please revert back to the 27 series. Change is the only constant, but bad GUI can be fixed!

CM630, 2024-07-11:
> Try to remember, people worked really hard on this change. People that clearly want to make PrusaSlicer better.
I have nothing against this “improvement”, as long as I can switch to the previous view.

mrwizardtx, 2024-07-11:
As a code smith myself, we all work hard coding, but when you have
circulated such a radical and yet not positive change to a release, it
shows ineptitude to the process. Alpha, beta release. Never shove it and
see if it works. We have a process, it was not followed in this case.

paulcobbaut, 2024-07-11:
Agreed.
The amount of work is irrelevant when 35 year old habits on a working system (a menu accessible via Alt-E, Alt-F, ...) are suddenly rendered useless. You don't just throw away something that has proven to be solid for decades.

neclimdul, 2024-07-11:
While the new menu implementation is a bit weird, its actually not that different from Firefox, Chrome, Edge, MS Office apps, and OrcaSlicer to name a few that hide the main menu. Weirdness aside, its actually a very common modern design, probably stemming unifying application designs across operations systems and the fact that Mac "hides" it in the menu bar rather then including it in the application.

If that's the intention, it just needs more work rather then a reverting.

draeath, 2024-07-12:
I am not happy about this change, and I won't be upgrading until it's reverted or we're provided with a suitable setting or other workaround.

----

> its actually not that different from Firefox, Chrome, Edge, MS Office apps, and OrcaSlicer to name a few that hide the main menu

This doesn't mean it's a good idea there, either.

paulcobbaut, 2024-07-12:
> While the new menu implementation is a bit weird, its actually not that different from Firefox, Chrome, Edge, MS Office apps, and OrcaSlicer to name a few that hide the main menu.

Alt-f opens the File menu in Firefox and Chrome. Alt-e opens the edit menu in Firefox and the menu in Chrome. OrcaSlicer has a directly accessible File menu. None of these options exist in PrusaSlicer 2.8.0, instead it threw 35 years of muscle memory down the drain.

(Tested today on Debian, I never used the other two programs you mention but would be surprised if alt-f and alt-e to open the menu are disabled in MS Office.)

pedjas, 2024-07-14:
Return normal menu bar please, or at least allow option so user who prefer standard menu bar can chose to have it.

Hiding menu behind buttons is plainly wrong. Adds necessary click to most often used options. It is really annoying.

mixer3d, 2024-07-14:
ok, I finally found a thread dedicated to this regression, and I would also like to ask to reconsider the "new interface", or have a choice. Using the slicer now is much more annoying, I don't need to waste the screen on big buttons and click more than necessary. In addition, gcode rendering/preview on older computers is terribly slow.

mixer3d, 2024-07-14:
https://www.infoworld.com/article/2337080/the-decline-of-the-user-interface.html

m00dawg, 2024-07-18:
Related: https://github.com/prusa3d/PrusaSlicer/issues/13000

m00dawg, 2024-07-18:
Also, as part of the new UI change, it looks like tabbing over when punching in rotations or scale factors no longer works

mruskai, 2024-07-26:
Just adding my voice to this UI design problem.  The trend of removing menu bars or hiding them away is a bad one.  It hurts productivity, usability, and intuitiveness.  The change should either be reverted, or an option added to preserver the sane menu layout.

KlausKraemer, 2024-07-26:
> > Try to remember, people worked really hard on this change. People that clearly want to make PrusaSlicer better.
> > I have nothing against this “improvement”, as long as I can switch to the previous view.

I won't deny that. But sometimes going for fashion isn't really worth the effort, as it is with the alteration of tthe menu. Making prctically all user's workflow worse is actually annoying. Good will does not always end with good results. The menu issue is a 100% proof for that.

backbonemedia, 2024-07-31:
What is it with UI developers these days, do they actually think we all use 28" 4K displays or touch devices operated by clumsy fingers that need large spacing? No, we are still using that 10 year old machine with mouse and keyboard, that still gets the job done fast.

You managed to fill the menu space on my screen with 7 items where there used to be 13. Thats a waste of pixels.
You made the tool icons smaller (36px -> 32px) while making the toolbars wider (50px -> 58px).
These 8px add to the 3px for the larger menu bar and make 13px less vertical space in the 3D viewer.

Do the users stop using a software if it does not stick to the newest design trends? No. It does not need to be stylish, it needs to stick to proven design patterns the users are comfortable with.

Please do not try to reinvent the UI. And if you do (as we see in 2.8), please keep a legacy/compact/mouse mode.
For now, I will stick with 2.7.

markusdd, 2024-08-04:
upvoting this as well. The new UI is absolutely horrible.

Hellreaver, 2024-08-06:
I really love seeing companies ruin products by 'modernizing' them. Aggravating. the new ui is aggravating.

KlausKraemer, 2024-08-07:
There are actually so many more important issues than working at the UI. Giving this any focus is already unintelligible, but even worsening a well functioning UI is  annoying.

Has there been one issue opened, asking for doctoring the UI? I don't think so. But there are many, many open issues ...

nfingerle559, 2024-08-16:
I dislike the new UI as well, but I think the gesture is more deeply concerning than the UI itself. To me, this is a clear signal that their priority over at Prusa is broadening the appeal of their "ecosystem" amid intensifying competition in the 3D printer space, even if it is at the expense of existing users. I for one look at PrusaSlicer as a tool, and I prefer functionality to take precedence over every other conceivable design choice. Hinting at a "web 2.0" overhaul with these changes worries me.

fe60, 2024-09-03:
@elomech

> The operation of the PrusSlicer has slowed down with this change and has become more cumbersome.

@mixer3d 

> In addition, gcode rendering/preview on older computers is terribly slow.

There are a couple of issues addressing the slow rendering of the preview. #12937 is the oldest one I found.

lukasmatena, 2024-09-13:
Thank you all for the feedback. After going through it, we have decided to reintroduce the system menu in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.1-rc1). The new wider top bar is unchanged, because it is needed to create space for the login/avatar and it also makes the search function much more discoverable (and also consistent between plater and config tabs). The sliders in Preview are also unchanged - the need for new sliders was mostly driven by technical reasons and if there are some functionality related issues left (they are sometimes difficult to see due to various platform-dependent settings, quirks, etc), we will fix them (a lot was already improved since first 2.8.0 public alpha). After going through this (and not only this) thread, it seems that the most hated part of the changes is by far the hidden menu.

Looking back, the choice to release 2.8.0 in the state it was in was probably not the best one. We have made couple of clearly incorrect assumptions about how our users interact with the software. The feedback that many people provided showed that certain menu options are actually used much more frequently than assumed (even for functions accessible from elsewhere or by widely known keyboard shortcuts), and that people generally rely on the menu much more than expected. IF we ever decide (and I am not saying that we will) to go in some similar way, it will not be before finding intuitive placement for accessing the features elsewhere and the menu structure would be reorganized accordingly.

> Has there been one issue opened, asking for doctoring the UI?

One of the suggestions is in this very thread ([here](https://github.com/prusa3d/PrusaSlicer/issues/12943#issuecomment-2211777636)). In general, there were many requests that the UI of PrusaSlicer is outdated and that it should be modernized and that placement of some controls is strange. Github is not the only source of feedback, and the spectre of people discussing here is not sampling our user base uniformly. And people who are new to 3D printing, run PrusaSlicer, don't like it and go someplace else will usually not voice their opinion anywhere.
Finding the balance between "most people like it" and "people who are used to it do not get mad" is not simple. Comparing PrusaSlicer 2.0.0 and 2.8.1, there is a lot of minor UI changes (IMHO mostly for the better), but we were always careful not to move too much stuff around so existing users don't have to re-learn things. That's why e.g. the 3D scene/Preview switch is still at the bottom-left corner (where it is difficult to discover for new users), that the Tab structure is still as it was 10 years ago (although e.g. OrcaSlicer tabs may make more sense) etc. I am not trying to say that the hiding of the menu was a good thing to do (it wasn't), I am just saying that SOME changes in the UI in the long term are needed, because sticking to the UI structure created in 2012 (when most of the functions was not even there) may even discourage completely new users from using the software.

Thanks again for the feedback and as I have written in the release notes, it will influence our further UI-related decisions.

Closing.

closed 2024-09-13 by lukasmatena

Tupson444, 2024-09-14:
>Changing something that works for the sake of being "modern" is never a good idea. Being "dated" is not a valid criticism of anything.

I completely agree. Changing something only because it wasn't changed recently is not good reasoning (that is, the only reasoning is "modern feel", but that should never be a priority in practical products). We see this often in various products, and we're kind of used to it, but it doesn't mean it's not a bad thing.

>That's why e.g. the 3D scene/Preview switch is still at the bottom-left corner (where it is difficult to discover for new users)

Just my opinion, but I think this is something that should be worked on - yes, people who are used to it would not like change, but if it can be improved for new users, it should be considered. For example, in SuperSlicer, there is an option in the settings to have it in the tabs (as 3D view, Sliced preview ...), so everyone can choose what they like more.

Reorganizing things just to make it more "modern" (which is relative), without focusing on actually improving the workflow isn't suiting anyone (old or new users). I personally didn't mind 2.8.0, because I rarely use the menu, but I do agree that it is annoying to have an additional click.

Instead, things can be made looking more "fresh" by changing colors, making new icon style, etc.

carlosjln, 2024-09-15:
**This issue is marked as closed, but not quite resolved.**

- "Log in" menu on the top right could be an item on the main menu next to configuration.
- move the search bar to the left of "expert mode"
- remove the useless "settings" button and whole menu bar.

[image]

**If PrusaSlicer team really want to improve UX, then:**
- prevent this UI layer shift every time a parameter is changed
- retain "Sliced Info" container and while there's a slicing in progress display the value "Processing..."
- it would be REALY handy if the user could copy the Sliced Info, either allow text selection and/or add a clipboard button to copy the whole content.

[image]

[image]

KimmoHop, 2024-10-22:
> widely known keyboard shortcuts

There are? Are they :)
If not counting _universals_ (copy, paste, undo, redo (this seems to be different in linux applications!)) I don't think I know any :/

For a right-hand-mouse two-fingers-type person (sample A: me) shortcuts need to be global (*i*talic, **b**old), understandable (D for drop), easy and used often (D for drop) - and preferably on the left side of the keyboard (50% or more will have different opinion on this). I do remember old phone number from 30 years ago and PI with 8 decimals (my two first computers didn't know it, so I had to), just don't count on new stuff - that's not advertising itself - to stick :D

paulcobbaut, 2024-10-23:
> There are? Are they :) If not counting _universals_ (copy, paste, undo, redo (this seems to be different in linux applications!)) I don't think I know any :/
>
> For a right-hand-mouse two-fingers-type [...]

There are still a lot of programs that advertise (alt-e) in Edit menu and (alt-f) in File menu by underlining the relevant letter in the menu interface. And Ctrl-s (save) and Ctrl-o (open), and many more, are written behind the actions in the menu (in many programs). So it is still advertised, and as far as I know largely the same since DOS and OS/2 on current Windows and Linux.

I don't understand why the restyling of the interface needs to disable these shortcuts. For this reason I still use PrusaSlicer 2.7.

Hint (I know you didn't ask for it): Learn to type blindly with ten fingers, it is fun and saves a lot of time.