prusa3d/PrusaSlicer #12944: Sliced view scrolling needs attention.

eneene, 2024-06-28:
### Description of the bug

Not really a bug. But the scrolling in the layer view used to be a lot better.
I am not sure if this issue is particular to diferent mice models, but it used to be that for each click of my scrolling wheel, the layer height bar would jump a single layer. It used to give great precision when analising slicers work. 
Now it it takes a lot more clicks for layer, making it feel sluggish and slow.

### Project file & How to reproduce

Just slice any model and navigate the layers bar on the right of the sliced window.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.0+MacOS-arm64

### Operating system

macos sonoma 14.5 (23F79)

### Printer model

irrelevant to the issue.

sp847242, 2024-07-11:
I usually use the keyboard arrow keys for this purpose, but I also just tested scrollwheel performance, and I'm seeing the same thing as you:

**v2.7.4**: Setting my scrollwheel to free-spin and spinning it quickly, I still get very smooth visual performance, with a view of every layer in the Gcode preview. If I keep spinning the wheel, on a test model, it goes through about 500 layers in 10-11 seconds, and every individual layer is shown. So it seems like this version did something to ensure that every single layer was displayed at least once.
**v2.8.0**: Effectively a very low framerate, with many skipped layers. If I had to guess, it's updating the preview at maybe 4-7fps. Doing that same scrollwheel test, the layer slider scrolled all the way between layers 1 and 500 in under 4 seconds, but I would estimate that the preview only updated about 20 times during that time, instead of 500. 

Probably related: Arrow keys also are no longer usable for quickly scanning through a model's layers: pre-v2.8.0, holding down the up or down arrow key would quickly scroll through every single layer. It was a very rapid update rate, with each layer displayed in sequence. It worked well.
In v2.8.0, this is no longer possible. When I press and hold an arrow key, the Gcode layer view moves by just one single layer, and then it stops updating, especially on more complex models. The next visual update generally happens when I release the arrow key, at which point the display jumps by several layers, proportionate to how long the key has been pressed. 

I've also noticed that the preview is generally slower in 3d space, such as when simply moving around the sliced model in 3d space: Noticeably lower framerate in v2.8.0.

YuSanka, 2024-07-17:
@eneene , thanks for your report. 
SPE-2417

fe60, 2024-09-03:
Seems to be a duplicate of #12937

lukasmatena, 2024-09-12:
Should be fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Feedback is welcome.

sp847242, 2024-09-12:
I'm still seeing the same behavior. 
Video's attached.
https://github.com/user-attachments/assets/5b3a087b-47c0-42c3-ab03-83833c4f07e3

(I had to change the window size to get a video that was <10MB. This same behavior is present with or without the sidebar, and regardless of window size.)

What's happening in the video: I use the mouse to grab the plus-sign in the vertical slider, and then release the mouse button. Then I press and hold the up arrow key. For a short time, it scrolls upward smoothly, but then it starts to stutter. At "19.25 (128)," I'm still holding the up arrow key. The final display update occurs when I release the key. 

Not shown in the video, but if I were to continue holding the up arrow key even longer, the display does finally reach the top layer, but it can take quite a long time.
It doesn't stutter at the same spots each time either. I just tried another time, and it stalled out at around 16mm, and then about 22 seconds later, it jumped to the top layer.

I just tried this same process in v2.7.0, pressing-and-holding the up arrow key, and I get extremely smooth performance from start to finish.

j28CLY, 2026-09-18:
This is still present in 3.0.0-alpha11: I made a simple part that's just a large cylinder, about 65dia and 200mm tall, with 15% gyroid infill. 
I press and hold the up-arrow, or use the scrollwheel, to move up through the layers.
For the first 50-60% of the model, the update rate is very fast and smooth. But then, somewhat abruptly, the update rate slows down substantially. The behavior is as if there's some kind of memory buffer for some number of Gcode commands, and then that buffer is exceeded, so it has to access the .gcode file directly. (not to say that that's what it's doing, but that's what the slowdown behavior feels like)