prusa3d/PrusaSlicer #12947: GCode preview is not visible

arshakuni, 2024-06-28:
### Description of the bug

GCode preview is not visible
[image]

### Project file & How to reproduce

[160x160test2.zip](https://github.com/user-attachments/files/16033424/160x160test2.zip)
[2024-06-28_20-22-33.zip](https://github.com/user-attachments/files/16033426/2024-06-28_20-22-33.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

win10

### Printer model

Reborn 2

malowz, 2024-06-28:
there was 3 topic with this problem already, they are now this one:

https://github.com/prusa3d/PrusaSlicer/issues/12908

Jan-Soustruznik, 2024-08-09:
of closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/12908

closed 2024-08-09 by Jan-Soustruznik