prusa3d/PrusaSlicer #12950: PrusaSlicer Warning

epetit91, 2024-06-28:
### Description of the bug

When I run PrusaSlicer 2.8 on an iMac running macOS Sierra, I get a warning message.
[image]

### Project file & How to reproduce

[Empty Project.3mf.zip](https://github.com/user-attachments/files/16036343/Empty.Project.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+MacOS-x64

### Operating system

macOS 10.12.6 (Sierra)

### Printer model

Prusa MK4

TerPuc, 2024-10-14:
Hello @epetit91 , thank you for your report!

Issue created: SPE 2517

FidelCapo, 2024-12-16:
This issue should be fixed in version 2.9.0-beta1.
Closing.