prusa3d/PrusaSlicer #12951: Prusa Connect Tab goes black after a few hours and cannot be fixed with our restarting the application

johnkray, 2024-06-28:
### Description of the bug

Prusa Connect Tab goes black after a few hours and cannot be fixed with our restarting the application. 

[image]

### Project file & How to reproduce

Log in to Prusa Connect wait a few hours and check back and the page is blank. This is very repeatable for me.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

MacOS Sonoma 14.1.1 (23B81)

### Printer model

MK4, XL, SL1S

neclimdul, 2024-06-28:
I suspect I'm running into this two on linux but instead of a few hours, its a few minutes. I wonder if there's some sort of threading problem.

mix579, 2024-06-30:
Same here (macOS)

kubispe1, 2024-07-01:
Hi, this issue has been persisting for a long time. We are investigating it and need to solve it soon. Thank you for reporting it.

ChrisHeerschap, 2024-07-01:
Seeing the same on MacOS - just having the ability to force a reload of the page would be good. I just checked and it had happened again but I got the alert of a new version available and when I reloaded, all was good.

kubispe1, 2024-07-04:
SPE-2396 - We will try to fix it. 
Thanks

dlively1, 2024-07-17:
Also running into this issue on MacOS with the latest build.

I haven't been able to work around it and tried a few different things but happy to provide any debug info if that could be helpful to resolving the problem.

One interesting note is I see my printers available under the printer selector. The problem seems to manifest when clicking on the "Prusa Connect" tab, or more importantly when showing the printer selection for "Send to Connect".

Any updates or anything else I could try to resolve this issue?

feesta, 2024-08-27:
I also see this on Sonoma 14.4.1 where the Prusa Connect (v2.8) tab is blank besides the header. Opening Prusa Connect in the browser shows the correct printer status. 

Unexpectedly I can still send prints to the printer (MK4 w/ FW 6.1.2+7894) from PrusaSlicer from the Plater tab.

neclimdul, 2024-11-21:
tried latest alpha through flatpak and clicking around the printables table it started lagging, then froze completely the same way so this still seems to be an issue.

ChrisHeerschap, 2024-11-29:
2.9.0-alpha1 on Mac OS Sequoia 15.1.1 does not fix the issue. Downloaded it today and had PrusaSlicer idle for less than an hour and came back to a status I knew was wrong. Looked like it had stopped updating. Clicked around and got to the dreaded white Connect screen. F5 key does nothing. CMD-R does not reload.

ChrisHeerschap, 2024-12-06:
When the issue is happening, sometimes I'll get the pop-up in the bottom right "missing or invalid security token".

[image]

With 2.9.0-alpha1, _sometimes_ it will reconnect and start working when I switch to the Prusa Connect tab from another tab, but this seems random.

ChrisHeerschap, 2024-12-07:
Update - I was re-reading the release notes on 2.9.0-alpha1 and saw that the hot keys might not work and that there is a new web entry under the "View" menu, "Reload Web Content" - and this _does_ work. Deleted my previous comments about it not working in 2.9.0-alpha1.

speedysk1, 2025-03-07:
Same issue, Windows 11, version 2.9.0

closed 2026-09-29 by prusaslicer-bot[bot]