prusa3d/PrusaSlicer #12953: Big Efficiency Improvement for XL, different layer heights on same layer

sk8board, 2024-06-28:
### Problem Statement
Slicers that support multiple extruders lack a big efficiency by not supporting different layer heights within the same layer.

### Recommended solution
An example is to allow a fine nozzle and fine layer height for the perimeter, but use a large nozzle and large layer height for infill.

### Describe how it would work

If the slicer would allow infill with a large nozzle and a large layer height, this would drastically reduce print time, which would help justify the higher cost of a multiple extruder printer. 

The picture below shows infill with 0.4mm layer height for speed and a perimeter with 0.2mm layer height for good appearance.

[image]

BaconFase, 2024-06-29:
The bare minimum of what you're asking for has already been possible with the slicer. There are several small issues with it and the most obvious one is that you cannot use a wipe tower. Here's a guide to printing with multiple nozzle sizes: read [the OP](https://forum.prusa3d.com/forum/original-prusa-xl-tool-changer-general-discussion-announcements-and-releases/guide-how-to-print-with-multiple-nozzle-sizes/), then read [this update post](https://forum.prusa3d.com/forum/postid/701187/).

- Basically you just setup the hardware (nozzles) and software (Printers>Extruder#) as you'd want. 
- In Plater View, leave the set extruder for the object as "default". 
- Setup the Print Settings>Multiple Extruders to the tools you want to use. 
- Print Settings>Advanced and set all Extrusion widths to zeros.
- To do the infill thing you just have to set "Combine infill every:" to 2

[multinozzlesize.zip](https://github.com/user-attachments/files/16038407/multinozzlesize.zip)

[image]

Could/should the whole process be improved/streamlined - yes. But can it already do the thing you're asking for - also yes.

sk8board, 2024-07-07:
Thank you for this update. Unfortunately, from this video, the layer height remains the same between the different tool heads (extruders). 

Looks like Orca Slicer is trying to catch-up to Prusa Slicer with regard to multi-extruder. This make sense that they need to implement multi-extruder before different layer height for each extruder.

I will submit my request there, so they are aware of the need, but expect they cannot support at this time (which is why I submitted my request here since Prusa Slicer already supports multi-extruder).

closed 2026-09-29 by prusaslicer-bot[bot]