prusa3d/PrusaSlicer #12958: Cannot Load New Offline Profiles

Gunner087, 2024-06-28:
### Description of the bug

An error is thrown when attempting to load offline profiles.

### Project file & How to reproduce

Open PrusaSlicer, open the configuration assistant, go to the configuration sources section, uncheck Prusa FFF from the online sources, click "Load..." under local sources, select the vendor_indices.zip file, an error appears that says "Failed to extract local archive" followed by the path to the zip file.

Screenshot:
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

macOS Sonoma 14.5

### Printer model

--

lukasmatena, 2024-06-29:
You should give it the file that you downloaded. Not unzip it and give it just the `vendor_indices.zip`.

Gunner087, 2024-06-29:
Thanks, the problem is the "Open 'safe' files after downloading" option in Safari's settings. It automatically unzipped the downloaded file so I did not know there was a different zip file. I tried recompressing the download, but PrusaSlicer would not accept that. Once I turned off "Open 'safe' files after downloading" and used the downloaded zip file it worked.

[image]

Jan-Soustruznik, 2024-07-17:
Hello, @Gunner087 , It seems that ticket can be closed, right?

closed 2024-07-19 by Gunner087