prusa3d/PrusaSlicer #12959: Nozzle Clog Serviceability

DJ-Miller, 2024-06-29:
**Is your feature request related to a problem? Please describe.**
Cleaning clogged nozzles at a low Z height is difficult, especially within the first 20mm or so.

**Describe the solution you'd like**
Move the Z axis to the top or at least a consistent position for cleaning, and then when clicking on retry, it would return to the saved Z position to continue printing

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Additional context**
Add any other context or paint drawing about the feature request here.
When the nozzle clogs on a very low Z height, it would be much easier if the unit would position not just only at the X and Y limit, but to raise the Z to a manageable level (much higher Z) from which the use of the nozzle cleaning needles could be more accessible to the nozzle opening.

gusdleon, 2024-06-29:
That's related to firmware and not to the slicer
PS only generates G-code and your machine follows it, if something happens in the process of printing the PS can't help only the firmware can decide what to do.

DJ-Miller, 2024-07-01:
Then, if the firmware can move to the X and Y limit, can't it also capture current Z, and raise the Z for nozzle clog accessibility and then return to the saved Z level?

gusdleon, 2024-07-01:
I'm not sure and I don't have any machines with me at the moment but I think if you pause the print you can move the axis and o. Resume It will resume like nothing (I repeat, I don't remember is it es the case)

But thia repo is not related to prusa firmware, so this is not the place to discuss things related to it

closed 2026-09-29 by prusaslicer-bot[bot]