prusa3d/PrusaSlicer #12963: Constantly Logs Out

eurik72, 2024-06-29:
### Description of the bug

Constantly Logs out of account and logs in after close-open app a few times

### Project file & How to reproduce

Close app, open app, close app, open app  (loged out), close app, open app (loged in)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 11 Pro

### Printer model

Prusa mk3s+

ufanders, 2024-06-29:
I'm seeing similar behavior, periodic pop-up notifications within the 3D view that I am logged in. I'm running MacOS 14.5 on a Macbook M3 Max.

mlask, 2024-07-03:
Same on M3 Pro, MacOS 14.5 - "logged in" popup every few minutes, with a small lag when doing other stuff like rotating/changing layers.

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.