prusa3d/PrusaSlicer #12964: Auto speed not working

Dimarsio, 2024-06-29:
### Description of the bug

When set to zero all print moves and remove other speed limits and auto speed set 300, the actual speed is about 15

### Project file & How to reproduce

[bunny.zip](https://github.com/user-attachments/files/16039855/bunny.zip)
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Win 10x64 

### Printer model

any

nmaggioni, 2024-07-15:
I think I am facing this issue as well: when used with the Classic perimeter generator, autospeed seems to be severely limited. Switching to Arachne fixes that and brings back the expected values that v2.7.4 produced.

The only difference between these two projects is the perimeter generator setting:
[classic_perimeters.zip](https://github.com/user-attachments/files/16240399/classic_perimeters.zip)
[arachne_perimeters.zip](https://github.com/user-attachments/files/16240400/arachne_perimeters.zip)

aknezevi-work, 2025-05-16:
Arachne also has this issue. It, however, heavily depends on geometry.
I am using Variable Layer Height feature, which is very powerful, and first layer height is kind of a variable layer height.
However, having set speeds hard coded as mm/s when using Variable Layer Height is bad because print will either go too fast on some places or too slow when it could be printing faster.
Therefore, I want to use auto speed.
It is just not consistent with speed giving. It will unnecessarily slow down the print speed sometimes, depends really on the Variable Layer Height and the amount of times i press "smooth".
I have several projects, sometimes Variable Layer Height "solves" the issue and the print is faster, sometimes it makes the issue and the print is slower. Sometimes it makes no difference. Depends on the geometry, when it really should not so much.

[Auto_Speed_3.zip](https://github.com/user-attachments/files/19976094/Auto_Speed_3.zip)
[Auto_Speed_4.zip](https://github.com/user-attachments/files/19976093/Auto_Speed_4.zip)

[Auto_Speed.zip](https://github.com/user-attachments/files/19976106/Auto_Speed.zip)

closed 2026-09-29 by prusaslicer-bot[bot]