prusa3d/PrusaSlicer #12965: Organic Support crashes the slicer

radzimir, 2024-06-29:
### Description of the bug

Switsching to organic support on one particular model crashes the slicer. 
In attached 3mf file this is switched already on. 

General remark: it would be nice if crash during slicing would be caught and contained internaly within crashing UI. 

### Project file & How to reproduce

[Sitz_Klemme_1.zip](https://github.com/user-attachments/files/16040141/Sitz_Klemme_1.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Windows 10

### Printer model

Prusa XL dual head, firmware 5.1.2

Jan-Soustruznik, 2024-07-17:
Hello, @radzimir , We cannot evoke your reported problem with crash. Problem should be evoked instantly after slice of attahced 3MF file? Could you provide more info about the issue? OR Could you check if reported problem isnt caused by some HW/SW problem, RAM,CPU,SSD,HDD ... old drivers on your PC? Thank you for you time. Have a nice day

radzimir, 2024-07-24:
It is working now, probably I should restart computer bevore reporting.

closed 2024-07-24 by radzimir