prusa3d/PrusaSlicer #12968: Remove webkit as a mandatory build-time dependency

buhman, 2024-06-29:
**Is your feature request related to a problem? Please describe.**
I originally bought my Prusa printers specifically because, unlike other vendors, there was no "cloud" feature.

It is unreasonable that Webkit is a mandatory build-time/run-time dependency, and it should be possible to compile PrusaSlicer such that all mention of "Prusa Connect" is removed entirely.

**Describe the solution you'd like**
Add a new Cmake option, for example `SLIC3R_WEBKIT` and/or `SLIC3R_PRUSACONNECT`. When cmake is given with an invocation such as `cmake -DSLIC3R_WEBKIT=OFF`, the compiled binaries should have no link/load dependencies on Webkit nor a wxGTK that has webview compiled-in. All buttons and menu options that refer to "Prusa Connect" should be removed if these options are OFF.

**Describe alternatives you've considered**
Stay on 2.7.4 forever.

gordon-quad, 2024-07-04:
Not sure what WebView is used for, but if there is an option to just turn off this functionality in full to make wxGTK webkit dependency optional - it would be great.

l29ah, 2024-07-04:
2276cedc107d3fd8cfc46d5f05cfe1d663b6b152 first bad commit

Perlovka, 2024-07-09:
+10
Webkit also has ruby requirement, pulling a lot of unneeded crap as dependencies.
```
Dependency resolution took 3.52 s (backtrack: 0/20).

[ebuild  N     ] dev-util/unifdef-2.12-r1::gentoo  66 KiB
[ebuild  N     ] app-eselect/eselect-ruby-20231226::gentoo  2 KiB
[ebuild  N     ] dev-libs/hyphen-2.8.8-r2::gentoo  USE="-static-libs" 0 KiB
[ebuild  N     ] media-libs/woff2-1.0.2-r6::gentoo  39 KiB
[ebuild  N     ] sys-apps/xdg-dbus-proxy-0.1.5::gentoo  USE="-test" 122 KiB
[ebuild  N     ] dev-util/colm-0.14.7-r3::gentoo  USE="-doc" 3,038 KiB
[ebuild  N     ] dev-util/ragel-7.0.4-r3::gentoo  USE="-doc" 1,618 KiB
[ebuild  N     ] media-plugins/gst-plugins-opus-1.22.11:1.0::gentoo  ABI_X86="(64) -32 (-x32)" 0 KiB
[ebuild  N     ] dev-lang/ruby-3.1.6:3.1::gentoo  USE="gdbm ipv6 ssl -berkdb -debug -doc -examples (-jemalloc) -jit -socks5 (-static-libs) -systemtap -tk -valgrind -xemacs" 14,916 KiB
[ebuild  N     ] dev-lang/ruby-3.2.4:3.2::gentoo  USE="gdbm ipv6 ssl -berkdb -debug -doc -examples (-jemalloc) -jit -socks5 (-static-libs) -systemtap -tk -valgrind -xemacs" 14,820 KiB
[ebuild  N     ] virtual/ruby-ssl-14::gentoo  RUBY_TARGETS="ruby31 ruby32 -ruby33" 0 KiB
[ebuild  N     ] dev-ruby/rubygems-3.5.11::gentoo  USE="-server -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 13,225 KiB
[ebuild  N     ] virtual/rubygems-20::gentoo  RUBY_TARGETS="ruby31 ruby32 -ruby33" 0 KiB
[ebuild  N     ] dev-ruby/power_assert-2.0.3::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 17 KiB
[ebuild  N     ] dev-ruby/test-unit-3.6.2:2::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 162 KiB
[ebuild  N     ] dev-ruby/bundler-2.5.11:2::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 423 KiB
[ebuild  N     ] dev-ruby/did_you_mean-1.6.3:2.6::gentoo  USE="-test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 46 KiB
[ebuild  N     ] dev-ruby/rake-13.2.1::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 122 KiB
[ebuild  N     ] dev-ruby/json-2.7.2-r1:2::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 123 KiB
[ebuild  N     ] dev-ruby/minitest-5.24.1:5::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 98 KiB
[ebuild  N     ] dev-ruby/matrix-0.4.2::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 42 KiB
[ebuild  N     ] dev-ruby/abbrev-0.1.2::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 5 KiB
[ebuild  N     ] dev-ruby/rbs-3.5.1::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 1,224 KiB
[ebuild  N     ] dev-ruby/typeprof-0.21.11::gentoo  USE="-test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 1,012 KiB
[ebuild  N     ] dev-ruby/strscan-3.1.0:3::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 27 KiB
[ebuild  N     ] dev-ruby/rexml-3.3.1:3::gentoo  USE="-test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 452 KiB
[ebuild  N     ] dev-ruby/rss-0.3.0::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 105 KiB
[ebuild  N     ] dev-ruby/date-3.3.4::gentoo  USE="-test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 97 KiB
[ebuild  N     ] dev-ruby/kpeg-1.3.3:1::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 57 KiB
[ebuild  N     ] dev-ruby/racc-1.8.0::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 447 KiB
[ebuild  N     ] dev-ruby/stringio-3.1.1::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 38 KiB
[ebuild  N     ] dev-ruby/psych-5.1.2:5::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 83 KiB
[ebuild  N     ] dev-ruby/rdoc-6.7.0::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 750 KiB
[ebuild  N     ] dev-ruby/io-console-0.7.2::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 37 KiB
[ebuild  N     ] dev-ruby/reline-0.5.9::gentoo  USE="-test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 91 KiB
[ebuild  N     ] dev-ruby/irb-1.14.0::gentoo  USE="-test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 138 KiB
[ebuild  N     ] dev-ruby/debug-1.9.2:1::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 164 KiB
[ebuild  N     ] dev-ruby/forwardable-1.3.3::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 18 KiB
[ebuild  N     ] dev-ruby/singleton-0.2.0::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 6 KiB
[ebuild  N     ] dev-ruby/prime-0.1.2-r1::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 10 KiB
[ebuild  N     ] dev-ruby/time-0.3.0::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 16 KiB
[ebuild  N     ] dev-ruby/timeout-0.4.1::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 8 KiB
[ebuild  N     ] dev-ruby/net-protocol-0.2.2::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 8 KiB
[ebuild  N     ] dev-ruby/net-ftp-0.3.7::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 30 KiB
[ebuild  N     ] dev-ruby/net-smtp-0.5.0::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 26 KiB
[ebuild  N     ] dev-ruby/net-imap-0.4.14:0.4::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 246 KiB
[ebuild  N     ] dev-ruby/net-pop-0.1.2::gentoo  USE="-doc -test" RUBY_TARGETS="ruby31 ruby32 -ruby33" 22 KiB
[ebuild  N     ] net-libs/webkit-gtk-2.44.1:4/37::gentoo  USE="X gstreamer introspection jumbo-build (seccomp) (-aqua) -avif -examples -gamepad -jpegxl -keyring -lcms -pdf -spell -systemd -wayland" 35,431 KiB
[ebuild   R    ] x11-libs/wxGTK-3.2.2.1-r4:3.2-gtk3::gentoo  USE="X gstreamer libnotify lzma opengl sdl spell tiff webkit* -curl -debug -doc -keyring (-pch) -test -wayland" ABI_X86="(64) -32 (-x32)" 0 KiB
[ebuild     U  ] media-gfx/prusaslicer-2.8.0::gentoo [2.7.4::gentoo] USE="-test" 68,171 KiB
```
I'll better stay on 2.7

luke-jr, 2024-08-12:
WebKit, like all HTML/JS engines, is prone to regular security vulnerabilities, and therefore banned from most of my machines. +1 to removing the mandatory dependency.

toomanydev, 2024-08-18:
I'm not using Prusa Connect or anything, but I killed all the webkit instances and the slicer still works fine. It would also be nice to see these instances to only be spawned if the user has done something that requires them (logging in, clicking the login button, etc).

SparkyBluefang, 2024-08-18:
+1 for making the webkit components optional. Though I approach this from a slightly different perspective. For source based Linux distros (Gentoo family, LFS, etc), pulling in webkit is a big compile time commitment which is exacerbated by semi-frequent security updates.

Admittedly, compile time is little more than an annoyance, but it _is_ annoying to add hours of time to regular system updates, especially for a feature not being used. The security angle is probably more compelling, since there are unavoidable considerations of embedding web technologies and loading external content.

I did poke around in the code (it has been nearly 20 years since I've touched cmake, and I'm not a c[++] dev by trade), and it looks like this is sort of a tangled mess, and it doesn't look like it will be straightforward to `#ifdef` away. IMHO the complexity is kind of disappointing, considering webkit already seems like the sledge hammer solution opposed to implementing a native API based integration.

PointyFluff, 2024-09-02:
The new webkit UI is incredibly buggy and nearly useless. I have to literally restart the app after every sliced model, or it stops updating the UI.    webkit needs to go.

jiri-jirus, 2024-10-07:
why is noone solving this problem? prusa-slicer is starting to become worst slicer of all from performance point of view and developers/maintainers are dead silent about this problem. CPU usage when doing nothing has always to be 0%

TerPuc, 2024-10-10:
Hello @buhman, thank you for your report!

Issue created: SPE 2512

cynicalcervine, 2024-10-12:
> why is noone solving this problem? prusa-slicer is starting to become worst slicer of all from performance point of view and developers/maintainers are dead silent about this problem. CPU usage when doing nothing has always to be 0%

I wrote a patch to just #if all of the webview code out a couple weeks ago that works well enough and just breaks the cloud functionality. Will take a bit more work to put all of it behind a feature gate and hide all the cloud print stuff, it's a bit messy, but it can be done.

gordon-quad, 2024-10-27:
@cynicalcervine where I can find your patch?

cynicalcervine, 2024-10-31:
@gordon-quad just uploaded into the `optional_webkit` branch on my fork. Do note that to get it to compile I had to use the Gentoo patchset and one additional patch to resolve a `GTK/GTK.h not found` error
```
--- a/src/slic3r/CMakeLists.txt
+++ b/src/slic3r/CMakeLists.txt
@@ -436,10 +479,15 @@ endif ()
 # link these libraries.
 if (UNIX AND NOT APPLE)
       find_package(GTK${SLIC3R_GTK} REQUIRED)
 +     target_include_directories(libslic3r_gui PRIVATE ${GTK${SLIC3R_GTK}_INCLUDE_DIRS})
 ```

buhman, 2024-10-31:
@cynicalcervine your commit looks pretty good. Do you think it's high enough quality to submit as a pull request here?

cynicalcervine, 2024-11-01:
@buhman maybe, but they'd probably want a bit more work done on it to make it so that none of the online functionality is enabled at all when you turn off the feature as right now it will show a login box that just does nothing. It would just involve a couple more #ifdefs sprinkled in where necessary.

l29ah, 2024-12-24:
I've upgraded prusaslicer to 2.8.1 and now it refuses to even start when it doesn't have access to launch the browser:
```
** (prusa-slicer:8423): ERROR **: 23:17:38.760: Unable to spawn a new child process: Failed to spawn child process “/usr/libexec/webkit2gtk-4.1/WebKitNetworkProcess” (Permission denied)

Thread 1 "slic3r_main" received signal SIGTRAP, Trace/breakpoint trap.
0x00007f0a071181d7 in g_log_structured_array () from /usr/lib64/libglib-2.0.so.0
(gdb) bt
#0  0x00007f0a071181d7 in g_log_structured_array () at /usr/lib64/libglib-2.0.so.0
#1  0x00007f0a07118637 in g_log_default_handler () at /usr/lib64/libglib-2.0.so.0
#2  0x00007f0a07118950 in g_logv () at /usr/lib64/libglib-2.0.so.0
#3  0x00007f0a07118d8b in g_log () at /usr/lib64/libglib-2.0.so.0
#4  0x00007f0a030be1e4 in WebKit::ProcessLauncher::launchProcess() () at /usr/lib64/libwebkit2gtk-4.1.so.0
#5  0x00007f0a02f1681e in WebKit::AuxiliaryProcessProxy::connect() () at /usr/lib64/libwebkit2gtk-4.1.so.0
#6  0x00007f0a030dbf78 in WebKit::NetworkProcessProxy::NetworkProcessProxy() () at /usr/lib64/libwebkit2gtk-4.1.so.0
#7  0x00007f0a030eafc5 in WebKit::WebsiteDataStore::networkProcess() () at /usr/lib64/libwebkit2gtk-4.1.so.0
#8  0x00007f0a030fce9a in WebKit::WebsiteDataStore::setIgnoreTLSErrors(bool) () at /usr/lib64/libwebkit2gtk-4.1.so.0
#9  0x00007f0a0306efcd in webkitWebsiteDataManagerGetDataStore(_WebKitWebsiteDataManager*) () at /usr/lib64/libwebkit2gtk-4.1.so.0
#10 0x00007f0a03065f12 in webkitWebViewConstructed(_GObject*) () at /usr/lib64/libwebkit2gtk-4.1.so.0
#11 0x00007f0a07f02606 in g_object_new_internal.part () at /usr/lib64/libgobject-2.0.so.0
#12 0x00007f0a07f0448e in g_object_new_with_properties () at /usr/lib64/libgobject-2.0.so.0
#13 0x00007f0a07f05409 in g_object_new () at /usr/lib64/libgobject-2.0.so.0
#14 0x00007f0a08dda871 in wxWebViewWebKit::Create(wxWindow*, int, wxString const&, wxPoint const&, wxSize const&, long, wxString const&) () at /usr/lib64/libwx_gtk3u_webview-3.2-gtk3.so.0
#15 0x0000561df043cb9a in WebView::CreateWebView (parent=parent@entry=0x561e116b9610, url=..., message_handlers=std::vector of length 1, capacity 1 = {...})
    at /var/tmp/portage/media-gfx/prusaslicer-2.8.1-r1/work/PrusaSlicer-version_2.8.1/src/slic3r/GUI/WebView.cpp:37
#16 0x0000561df0438a85 in Slic3r::GUI::WebViewPanel::WebViewPanel
    (this=this@entry=0x561e116b9610, parent=parent@entry=0x561e0f5798e0, default_url=..., message_handler_names=std::vector of length 1, capacity 1 = {...}, loading_html="connect_loading")
    at /var/tmp/portage/media-gfx/prusaslicer-2.8.1-r1/work/PrusaSlicer-version_2.8.1/src/slic3r/GUI/WebViewDialog.cpp:112
#17 0x0000561df0438fa4 in Slic3r::GUI::ConnectWebViewPanel::ConnectWebViewPanel (this=this@entry=0x561e116b9610, parent=0x561e0f5798e0)
    at /var/tmp/portage/media-gfx/prusaslicer-2.8.1-r1/work/PrusaSlicer-version_2.8.1/src/slic3r/GUI/WebViewDialog.cpp:596
#18 0x0000561defffe71f in Slic3r::GUI::MainFrame::create_preset_tabs (this=0x561e0f87c6e0) at /var/tmp/portage/media-gfx/prusaslicer-2.8.1-r1/work/PrusaSlicer-version_2.8.1/src/slic3r/GUI/MainFrame.cpp:804
#19 0x0000561defffee80 in Slic3r::GUI::MainFrame::init_tabpanel (this=this@entry=0x561e0f87c6e0) at /var/tmp/portage/media-gfx/prusaslicer-2.8.1-r1/work/PrusaSlicer-version_2.8.1/src/slic3r/GUI/MainFrame.cpp:722
#20 0x0000561df0002828 in Slic3r::GUI::MainFrame::MainFrame (this=this@entry=0x561e0f87c6e0, font_point_size=<optimized out>)
    at /var/tmp/portage/media-gfx/prusaslicer-2.8.1-r1/work/PrusaSlicer-version_2.8.1/src/slic3r/GUI/MainFrame.cpp:179
#21 0x0000561deffd16f0 in Slic3r::GUI::GUI_App::on_init_inner (this=<optimized out>) at /var/tmp/portage/media-gfx/prusaslicer-2.8.1-r1/work/PrusaSlicer-version_2.8.1/src/slic3r/GUI/GUI_App.cpp:1501
#22 0x0000561deffd314d in Slic3r::GUI::GUI_App::OnInit (this=<optimized out>) at /var/tmp/portage/media-gfx/prusaslicer-2.8.1-r1/work/PrusaSlicer-version_2.8.1/src/slic3r/GUI/GUI_App.cpp:1170
#23 0x00007f0a08b2ae72 in wxEntry(int&, wchar_t**) () at /usr/lib64/libwx_baseu-3.2-gtk3.so.0
#24 0x0000561deffa8ac8 in Slic3r::GUI::GUI_Run (params=...) at /var/tmp/portage/media-gfx/prusaslicer-2.8.1-r1/work/PrusaSlicer-version_2.8.1/src/slic3r/GUI/GUI_Init.cpp:71
#25 0x0000561def50e1b1 in Slic3r::CLI::run (this=this@entry=0x7ffc8f985460, argc=<optimized out>, argv=<optimized out>) at /var/tmp/portage/media-gfx/prusaslicer-2.8.1-r1/work/PrusaSlicer-version_2.8.1/src/PrusaSlicer.cpp:764
#26 0x0000561def4e006f in main (argc=<optimized out>, argv=<optimized out>) at /var/tmp/portage/media-gfx/prusaslicer-2.8.1-r1/work/PrusaSlicer-version_2.8.1/src/PrusaSlicer.cpp:1159
```

luke-jr, 2024-12-29:
@cynicalcervine Sadly, your patch does not work with 2.8.1.

sithlord48, 2025-01-01:
+1 Local only, Web browsers are a thing YOUR better then them stop trying to be them

codefaux, 2025-01-01:
Installing a web browser inside your 3D model slicer is insane.

I have a web browser, if I need it I'll use it, I promise.

Is it within scope to add a compiler and infrastructure to download, configure and compile firmware?
Why not bundle in a bunch of search platform toolbars instead, like the good old days? I feel like they pay better.
Why not build a kernel and make it bootable, compile in Xorg and GCC?
Can I use the slicer to make coffee and a bagel yet?
The spell checker built into the text tool isn't very good, better add OpenOffice.
HEY why not add AI? Pytorch is only a few GBs, a good conversational model is only a few more GNs, nobody will mind, and most people have modest GPUs right?

Seriously. It's a slicer. It doesn't need a food processor attachment. Stop spending your time on the shoe polisher. This is beyond feature creep.

If this isn't addressed, I'll stop using PrusaSlicer. I haven't used another slicer since Skeinforge, and if I leave I won't look back. No big loss, but I'm not alone.

Most of the compelling features in this slicer have come from outside this project recently anyway.

CRCinAU, 2025-01-01:
Agreed. This added a hell of a lot of issues with distribution, building, compiling and operation of a slicer.

Rip out the web browser completely from PrusaSlicer and bring it back to what it actually is - a slicer.

sithlord48, 2025-01-01:
If you must keep the tabs for printables and local printer how about just having them open in my browser. Tons of other apps do this for opening web content. This is even how most of the item in the help menu work already so the code is even here todo it. 

The only "web" thing i want to be able todo from the slicer is upload gcode to my printer.

luke-jr, 2025-01-01:
> @cynicalcervine Sadly, your patch does not work with 2.8.1.

Looks like this is a bug in unpatched 2.8.1. Fixed with:

```
--- a/src/slic3r/CMakeLists.txt
+++ b/src/slic3r/CMakeLists.txt
@@ -487,6 +487,8 @@
     else ()
         target_link_libraries(libslic3r_gui PUBLIC ${GTK${SLIC3R_GTK}_LIBRARIES} fontconfig)
     endif ()
+    include_directories(${GTK${SLIC3R_GTK}_INCLUDE_DIRS})
+    add_definitions(${GTK${SLIC3R_GTK}_CFLAGS_OTHER})
 endif ()
 
 # Add a definition so that we can tell we are compiling slic3r.
```

strayr, 2025-01-06:
Can we please, please, please have at least an options level toggle to NOT have printables and physical printer tabs in the slicer. My web browser is very good at managing system-hog pages that I'm not currently engaged with. I'm quite happy to alt-tab between prusaslicer and and a web browser and close the browser when i don't need to poke my printer. I also have about 20 printables tabs open.

I don't need that in my slicer. I just need my slicer to not fall over on my i7-8565U laptop with only 16GiB of ram. It should be plenty for this.

A "fix" for kilpper users is to specify the moonraker port, so I set `fastprinter.local:7125` as the URL and the physical printer page shows a very minimal moonraker info page.

codefaux, 2025-01-06:
@cynicalcervine @luke-jr -- they closed #13635 as "completed" because they fully intend to continue to chase profits at the expense of an acknowledged but unvalued portion of their userbase. Are either of you going to / willing to keep a repo up to date for this? If I need to change slicers, I'd rather start now.

codefaux, 2025-01-06:
> Can we please, please, please have at least an options level toggle to NOT have printables and physical printer tabs in the slicer.

@strayr -- The answer is a definitive hard no. #13635 was just closed, asking for specifically that.

> I don't need that in my slicer. I just need my slicer to not fall over on my i7-8565U laptop with only 16GiB of ram. It should be plenty for this.

But what about PrusaSlicer's profits? Those poor profits.

> A "fix" for kilpper users is to specify the moonraker port, so I set `fastprinter.local:7125` as the URL and the physical printer page shows a very minimal moonraker info page.

Thank you for this, I'll be implementing it until I find a replacement slicer.

Does anyone have any suggestions? I'm guessing OrcaSlicer is where PrusaSlicer got the idiot idea of integrating a browser in the first place, what else is there?

strayr, 2025-01-06:
I didn't ask about disabling downloads, getting an STL straight to PS is
cool. I just want the web browser in my slicer gone. It's an absolute
performance killer. And all it takes is one mis-click and it's there.

I mean I could block printables on my local net as I usually browse through
a VPN, but that's a little extreme.

I'd be sticking with 2.7 or an ancient build of super slicer if I was still
on a really feeble machine. It really shouldn't be the case that PS uses
more resources than Fusion because someone thought it would be a good idea
to embed a web browser in it that I can't turn off.

Just let me turn it off at runtime. Please.

CRCinAU, 2025-01-10:
Enshittification rarely is reversed once its started.

I would strongly suggest to the community to start looking at alternatives, or coordinate to maintain a fork.

codefaux, 2025-01-10:
> Sadly that's probably not something I'll do into the future. I'll likely update my fork if I really have to [...]  I'll probably just end up picking a different slicer myself when I really need it :(

Thanks for the update. I figured this was likely the case. Between Prusa and a few other apps recently shittifying themselves and the EOL/update pressure/shittifying direction of Win10, I recently abandoned it (NEVER Windows 11) and moved to main Linux. I'm forced to use the PrusaSlicer Flatpak while I look for a new solution, as I'm not prepared to just shut down in the mean time. The flatpak shits the bed and freezes any time I accidentally click either the Physical Printer or Printables tabs, and it can't access my /mnt/win folder where all the models live for now because it's a flatpak and not an actual application which means everyone ELSE gets to decide what permissions I have with it, so I'm beyond doubly driven to find a replacement. Do you have thoughts on candidates?

> Sad to see them doing this in such a sloppy way. It was like 2 hours of work to just put everything behind a build flag. It's not even that it would be super hard to maintain just this one patch, but it looks like they plan on integrating even more with web stuff going forward.

Yeah, that's pretty much the one. Their reply to the other read as "shareholders agree that this is the way moving forward but I can't say that out loud" and it just fully made me lose interest. The good features come from other slicers, the shit comes from above. The main thing they're worried with is digesting it and making the profit needle move.

There will soon be no Prusa community; they're viewed as just a pool of consumers.

l29ah, 2025-01-10:
Please tell if there's a PrusaSlicer ←→ SuperSlicer configuration directory converter.

CRCinAU, 2025-01-10:
For what its worth, the PrusaSlicer 2.7.5 AppImage works just fine in all respects.

There's very little added since that is a 'must have' to force an upgrade.

codefaux, 2025-01-10:
> For what its worth, the PrusaSlicer 2.7.5 AppImage works just fine in all respects.

Yeah thanks for the tip. I actually recently found out about AppImage with my CAD software. I'm impressed with it. It makes it __really__ obvious how much bloated BS you add to an app though, so I can see why PrusaSlicer would want to hide that set of holiday photos, so to speak.

Wait..did they __remove__ 2.7.5 from Github? All I see is 2.7.5-rc1 presently.

> There's very little added since that is a 'must have' to force an upgrade.

Disagree. Multiple bed / bulk export is a small bonus for me.

Scarf Seams has made a -HUGE- improvement in print visual quality. Came from OrcaSlicer, but OrcaSlicer embraces a lot of things I don't want too. No corporate [censored] going on, so that's a plus, but it's almost a lateral move in "unwanted extras" to me.

Equidamoid, 2025-01-31:
One more +1 to this.
Saving 2 clicks to download & open is not worth adding such a monstrous dependency and a potential security risk.

Runtime switch would be okay. Compile-time switch would be perfect.

git4rker, 2025-03-01:
I really don't like having to install WebKit on my system just to use my 3D printer.
Like, it's useless; AFAIK the only place where it's used is the Physical Printer tab, but whoever implemented it clearly didn't check how it performs. It doesn't even have hardware acceleration.

darsto, 2025-03-12:
Would it be possible to make a webkit stub and automate prusaslicer compilation? Hopefully this wouldn't require any changes to prusaslicer code - only some env variables when building.

codefaux, 2025-03-12:
> I really don't like having to install WebKit on my system just to use my 3D printer. Like, it's useless; AFAIK the only place where it's used is the Physical Printer tab, but whoever implemented it clearly didn't check how it performs. It doesn't even have hardware acceleration and is extremely buggy.

It is also used in the non-optional Printables integration, said integration being involved in a separate issue as users also take issue to the inclusion of a website-tie-in integration built into their slicer.

As to extremely buggy, I can agree, my slicer locks up the instant a page finishes loading when I accidentally click on either the Printables OR the Physical Printer tab. So, I have to carefully avoid my "it's the furthest right button" instinct and make really sure I don't mis-click, or my slicer crashes, because of an integration many people both object to and have issues with.

> Won't be surprised if PrusaSlicer will eventually become an Electron app. The enshittification process seems to have started.

Oh man I'd actually prefer that over this garbage integration hackjob they've pushed on us. Then the whole thing would either work or not work. Then it wouldn't be stuck in fking Flatpak so I can't access my mountpoints under /mnt to get to my prints. Then it wouldn't be both A) everything required to run a native app and B) everything required to run a web browser integration. IN WHAT USED TO BE A SLICER. Now I have to worry about webkit CVEs in my FSCKING SLICER! ARE YOU KIDDING ME.

..sorry, deep breath.

> Would it be possible to make a webkit stub and automate prusaslicer compilation? Hopefully this wouldn't require any changes to prusaslicer code - only some env variables when building.

It sure would! They definitely won't! It's the money. It's the integration with Printables. And if you don't know what it is, you won't know it's an external intgegration when you see it, so you'll accidentally or intentionally wind up there, making them that sweet money.

DiThi, 2025-03-20:
I lost several projects already because using any of those tabs crashes the application, when all I wanted is to touch something in the printer config (which used to be the rightmost tab)... If it's not working well it should be completely optional!

gordon-quad, 2025-03-21:
> I would strongly suggest to the community to start looking at alternatives, or coordinate to maintain a fork.

Well, looks like orcaslicer is plagued by the same webkit dependency and superslicer is dead, so we do not have working maintained slicer right now.

l29ah, 2025-03-21:
Fixed up and packaged the latest webkit-free prusaslicer release to build and work well on a modern system: https://github.com/l29ah/booboo/tree/master/media-gfx/prusaslicer

Perlovka, 2025-03-22:
@l29ah Latest non-webkit version is 2.7.4 and there is ebuild for it in the gentoo repo
https://github.com/gentoo/gentoo/blob/8d6478925ca0d8aa6b2c371ae0f06edac157cf01/media-gfx/prusaslicer/prusaslicer-2.7.4.ebuild

l29ah, 2025-04-07:
Thanks, updated.

codefaux, 2025-06-27:
This whole thing has shaken my 3D printing passion.

I can't make PrusaSlicer stable on Linux __using their Flatpak__ -- isn't that the point of Flatpak by the way? Making stable distribution easier? Nevermind that I can't access my files from inside the Flatpak and have to diddle around with the Flatpak or my own system in order to do it. That's annoying AF but pay no attention to advanced users I guess.

It's moving further from my interests. This project is vampiring ideas __and actual code__ from other open source projects and the community itself, and then adding broken bloat and integration against its users' best interests in pursuit of accidental/convenience-driven clickthrough to the parent website rather than __providing a good product people want to use the normal way__ and leaving the Slicer alone.

And I still have to use it. It still crashes if I accidentally trigger either webview tab which I don't want present in the first place. Can anyone suggest an alternative? I genuinely want to move away from this. I'm on Linux.

sithlord48, 2025-06-27:
This literally has stopped me from buying a 5 head XL and Core One (with mmu3)....

Perlovka, 2025-06-27:
Well, I've created yet another patch to remove this peace of bloatware from slicer.
I've built v2.9.2 on gentoo (with gentoo patches) without webkit and it still able least to start, slice and upload gcode to klipper host.

An ebuild and patches can be found here:
https://github.com/Perlovka/portage-overlay/tree/master/media-gfx/prusaslicer

Hope this will help somebody. Not tested windows and mac builds.

codefaux, 2025-06-27:
> This literally has stopped me from buying a 5 head XL and Core One (with mmu3)....

But but -- imagine how many handfuls of pennies they made by "conveniently" integrating a full (insecure, broken) web browser into the app and tieing it to their own service! Think of the poor CEO's private flight fueling costs! You can't make eternal returns that always increase from __selling printers and consumables__, you have to capture revenue from the __software__ too!

Seriously. I haven't even bought Prusa __filament__ since PrusaSlicer became the least stable part of my CAD -> 3D printing toolpath. It was pretty good stuff, too.

l29ah, 2025-06-27:
> > Well, I've created yet another patch to remove this peace of bloatware from slicer. I've built v2.9.2 on gentoo (with gentoo patches) without webkit and it still able at least to start, slice and upload gcode to klipper host.
> 
> You're incredible. Is there any chance you (or anyone else) could write this in a manner buildable on non-Gentoo?
> 
> A PKGBUILD for Arch-based distros would be my dream, or maybe just a compiled AppImage to reach more targets?

You can use the Gentoo Prefix feature on non-Gentoo systems.

Perlovka, 2025-06-27:
@codefaux never used arch :)
Luckily there are KVM images with preinstalled system.
But it fails to build because of OOM killer, even with 12 Gb RAM and 8 CPU and -j6
Can you try on your system?
https://github.com/Perlovka/arch-packages/tree/main/prusa-slicer

codefaux, 2025-06-27:
> > Arch is for real Linux users
> 
> Gentoo one love (20 years of love, actually. F**ck, i'm so old 😢 ) 🤗

I said I tried Gentoo two decades ago, and it wasn't my first Linux distro -- right there with you my friend.

> But it fails to build because of OOM killer, even with 12 Gb RAM and 8 CPU and -j6 Can you try on your system? https://github.com/Perlovka/arch-packages/tree/main/prusa-slicer
> 
> P.S. pay attention to ninja -j6 in PKGBUILD, maybe you need to increase it or remove -j completely, so it will build faster. P.P.S finally its succeeded
> 
> [image]

Drastically OT - moving to new issue (open Discussions?) on @Perlovka repo. (Builds, ish. Runs. Consider it, folks.)

synthead, 2026-08-15:
I built a package in the AUR to remove webkit from Prusa Slicer: https://aur.archlinux.org/packages/prusa-slicer-no-webkit

l29ah, 2026-08-16:
> I built a package in the AUR to remove webkit from Prusa Slicer: https://aur.archlinux.org/packages/prusa-slicer-no-webkit

Thanks a lot, adapted it for Gentoo and pushed to ::booboo.

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena