prusa3d/PrusaSlicer #12969: Both Physical Printer Tab and Prusa Connect Login Popup are Blank

mlgold, 2024-06-29:
### Description of the bug

On updated RHEL8, both the Physical Printer Tab and Prusa Connect Login Popup are Blank. They worked once, but after closing and opening Prusa Slicer again, they are blank. Prusa Link to the printer and Prusa Connect work on a browser.

### Project file & How to reproduce

Open Prusa Slicer. Click on Physical Printer Tab or the Login icon at top right.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Physical Printer Tab and Prusa Connect Login Popup are Blank

### Operating system

RHEL 8

### Printer model

Mini

PatB42, 2024-07-02:
Yes exactly that I was afraid , each time you close the slicer then you have to re write the name and password !! that why mozilla is best for such reminding.. and hope you find solution that not forced user to log in prusa acc !! for my all customer it's not good ..

mlgold, 2024-07-03:
> Yes exactly that I was afraid , each time you close the slicer then you have to re write the name and password !! that why mozilla is best for such reminding.. and hope you find solution that not forced user to log in prusa acc !! for my all customer it's not good ..

That is not my issue. I cannot enter the login password into the slicer at all, as the dialog box pops up blank and never fills in. The same happens with the Physical Printer tab. My guess is the Webkit integration is not working correctly. There are no error messages related to the popup or page selection, but lots of error messages when started on the command line.

`$ /usr/local/bin/PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929.AppImage

(prusa-slicer:75526): Gtk-WARNING **: 13:12:36.820: Theme parsing error: gtk-contained.css:2848:13: Expected a length

(prusa-slicer:75526): Gtk-WARNING **: 13:12:36.820: Theme parsing error: gtk-contained.css:2880:37: The :prelight pseudo-class is deprecated. Use :hover instead.

(prusa-slicer:75526): GLib-GObject-WARNING **: 13:12:37.473: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.473: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:75526): GLib-GObject-WARNING **: 13:12:37.473: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.473: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:75526): GLib-GObject-WARNING **: 13:12:37.474: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.474: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:75526): GLib-GObject-WARNING **: 13:12:37.475: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.475: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:75526): GLib-GObject-WARNING **: 13:12:37.476: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.476: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:75526): GLib-GObject-WARNING **: 13:12:37.478: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.478: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:75526): GLib-GObject-WARNING **: 13:12:37.481: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.481: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:75526): GLib-GObject-WARNING **: 13:12:37.481: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.481: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.497: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.497: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.500: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.500: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.703: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.703: gtk_widget_set_size_request: assertion 'width >= -1' failed

(WebKitWebProcess:75566): Gtk-WARNING **: 13:12:37.884: Theme parsing error: gtk-contained.css:2848:13: Expected a length

(WebKitWebProcess:75566): Gtk-WARNING **: 13:12:37.884: Theme parsing error: gtk-contained.css:2880:37: The :prelight pseudo-class is deprecated. Use :hover instead.

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.908: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.908: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.909: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.909: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:75526): Gtk-CRITICAL **: 13:12:37.910: gtk_widget_set_size_request: assertion 'width >= -1' failed

(WebKitWebProcess:75570): Gtk-WARNING **: 13:12:37.932: Theme parsing error: gtk-contained.css:2848:13: Expected a length

[... 87 lines cut ...]

(prusa-slicer:75526): Gtk-CRITICAL **: 13:13:49.815: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:75526): Gtk-CRITICAL **: 13:13:49.815: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:75526): GLib-GObject-WARNING **: 13:13:49.815: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:75526): Gtk-CRITICAL **: 13:13:49.815: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:75526): GLib-GObject-WARNING **: 13:13:49.815: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:75526): Gtk-CRITICAL **: 13:13:49.815: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
13:13:53: Debug: window wxTreeCtrl@0xc418b60 ("treeCtrl") lost focus even though it didn't have it
[... cut ...]
(WebKitWebProcess:76094): Gtk-WARNING **: 13:14:00.646: Theme parsing error: gtk-contained.css:2848:13: Expected a length

(WebKitWebProcess:76094): Gtk-WARNING **: 13:14:00.646: Theme parsing error: gtk-contained.css:2880:37: The :prelight pseudo-class is deprecated. Use :hover instead.

(WebKitWebProcess:76094): Gtk-WARNING **: 13:14:00.762: Theme parsing error: gtk-contained.css:2848:13: Expected a length

(WebKitWebProcess:76094): Gtk-WARNING **: 13:14:00.762: Theme parsing error: gtk-contained.css:2880:37: The :prelight pseudo-class is deprecated. Use :hover instead.
`

PatB42, 2024-07-03:
oulla are you sure is not missing 1 files !!!!!!!!!!!! haha so much.....

ThomasBoettcherHolschen, 2024-07-18:
Just installed the Slicer Version 2.8.0 via Flatpack on Linux Mint 21.3.
This is what happens when I click Login:

[image]

I can see changes in the Cursor when hovering the Form. I can click on something, but I don't know what happens ;-)
Let me know when you need more info.

snhirsch, 2024-07-23:
I'm having the same issue: Version 2.8.0+linux-x64-GTK3 running on Kubuntu (KDE desktop) 18.04.  Everything in the slicer works fine, but it doesn't show anything but a blank screen under 'Prusa Connect'.

grndkntrl, 2024-07-26:
Same problem on v2.8.0 on Win 11 x64:

[image]

I just updated from v2.6.4 which was fine,

Completely uninstalling & reinstalling from scratch doesn't fix it.

Rose-David, 2024-08-09:
Same issue on NixOS 24.11 Vicuna. I attached a log snippet of what happens when I attempt to open the login window.

```
failed to bind extensions
Failed to create GBM device for render device: /dev/dri/renderD128: No such file or directory
Failed to get GBM device
[2024-08-09 14:06:22.223435] [0x00007fa2257c1d80] [error]   WebViewDialog error: wxWEBVIEW_NAV_ERR_OTHER
Failed to get GBM device
Failed to get GBM device
Failed to get GBM device
```

Rose-David, 2024-08-11:
> I can see changes in the Cursor when hovering the Form.

I tried to see if I could interact with anything, by selecting all and dragging, and I got this:
[image]
It says "Connection failed, Something went wrong."

There doesn't appear to be anything else hidden as far as I can tell.

Rose-David, 2024-08-12:
So, @30350n found out that these invisible messages briefly tell you to install `glib-networking`, and once we rebuilt the package with that in the buildInputs, it worked flawlessly!

Can someone on another OS try this to see if it works?

snhirsch, 2024-08-13:
I'd love a fix for my Linux installation and would be interested in building here.  However, I'm not an expert on cmake and you have not provided quite enough detail for me to proceed.

30350n, 2024-08-13:
(In theory you shouldn't have to mess with cmake at all, but just install `glib-networking`. I'm guessing it's automatically detecting whether or not it's present.)

snhirsch, 2024-08-13:
I cannot build from source on Ubuntu 18.04 LTS.  The available Boost library is one minor version too old, apparently.   One of these days I'll upgrade my system...  For now it's the end of the line.

jmickelin, 2024-08-16:
> So, @30350n found out that these invisible messages briefly tell you to install `glib-networking`, and once we rebuilt the package with that in the buildInputs, it worked flawlessly!
> 
> Can someone on another OS try this to see if it works?

Curiously, I am also running NixOS, and encountered the bug even after your `nixpkgs` patch was merged.

I managed to solve it by disabling Webkit's hardware acceleration, done by running the program with `WEBKIT_DISABLE_COMPOSITING_MODE=1`. For reference, see this ticket for another program displaying the same issue: https://github.com/tauri-apps/tauri/issues/8254

```
$ WEBKIT_DISABLE_COMPOSITING_MODE=1 prusa-slicer 
```

Apparently, at least for one person in that thread, the problem was caused by a version mismatch between the system's Mesa libraries and those used by `webkitgtk`. Now, that might be completely NixOS specific (in my case, since the majority of my system is running on `24.05` and only a few programs, PrusaSlicer included, on `unstable`), but it wouldn't be entirely out of the question to investigate for other bundled releases, like Flatpak, mentioned above...

I'll have to get back to you after trying the more proper fix of building `webkitgtk` with the right Mesa version (also described in that other issue).

For reference, the stderr in my case has a few more error messages compared to the ones posted above:

<details>
<summary>PrusaSlicer output</summary>

```

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:10.759: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:10.760: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:10.764: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:10.766: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:10.769: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:10.770: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.032: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.032: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.034: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.034: gtk_widget_set_size_request: assertion 'width >= -1' failed
[2024-08-16 16:24:11.086052] [0x00007fcebd416d80] [error]   UserAccount: Failed to read token - no datafile found.

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.194: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.194: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.286: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.286: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.288: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.288: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.290: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.321: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.321: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.342: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.343: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.343: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:58033): Gtk-CRITICAL **: 16:24:11.345: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

[... 23 lines cut ...]
04:24:12 PM: Debug: window wxTreeCtrl@0x5594d309aeb0 ("treeCtrl") lost focus even though it didn't have it
04:24:12 PM: Debug: window wxTreeCtrl@0x5594d309aeb0 ("treeCtrl") lost focus even though it didn't have it
did not find extension DRI_Mesa version 2
did not find extension DRI_IMAGE_DRIVER version 2
failed to bind extensions
did not find extension DRI_Mesa version 2
did not find extension DRI_IMAGE_DRIVER version 2
failed to bind extensions
did not find extension DRI_Mesa version 2
did not find extension DRI_SWRast version 5
failed to bind extensions
Failed to create GBM device for render device: /dev/dri/renderD128: Illegal seek
did not find extension DRI_Mesa version 2
did not find extension DRI_IMAGE_DRIVER version 2
failed to bind extensions
did not find extension DRI_Mesa version 2
did not find extension DRI_IMAGE_DRIVER version 2
failed to bind extensions
did not find extension DRI_Mesa version 2
did not find extension DRI_SWRast version 5
failed to bind extensions
Failed to create GBM device for render device: /dev/dri/renderD128: Illegal seek
```

</details>

30350n, 2024-08-16:
> a version mismatch between the system's Mesa libraries and those used by `webkitgtk`
> only a few programs, PrusaSlicer included, on `unstable`

Yeah I also noticed this while testing the patch. You could probably get that setup to work though by using an `override` on the unstable `prusa-slicer` package to set the relevant GFX libraries to their stable version.

snhirsch, 2024-08-16:
That environment variable did not change the misbehavior on my system.  When I login to my Prusa account and select the Prusa Connect tab I get an immediate system notification that Webkit has crashed along with a blank window.

rkiddy, 2024-08-16:
Just FYI, I am able to build on Ubuntu 22.04.4 LTS. I actually pulled the source and built on Aug 13, in case that makes a difference. I was able to build with the directions, with no problems at all. And the Login window works. It is not empty.

snhirsch, 2024-08-16:
It doesn't even come close to building on Ubuntu 18.04 LTS.  Many pre-reqs missing and despite verbiage in the build instructions it makes no attempt to get them.  I'm hoping to install 24.04 LTS soon.

bkerler, 2024-09-20:
I can also confirm on Ubuntu 24.04 LTS that the window login and connect are blank. Running with WEBKIT_DISABLE_COMPOSITING_MODE=1 works, but only for regular deb and appimage for both 2.8.0 and 2.8.1.

bkerler, 2024-09-20:
I just figured out it is related to webkit using dma mode (especially for nvidia graphic cards). Running WEBKIT_DISABLE_DMABUF_RENDERER=1 seems to be the proper solution instead of WEBKIT_DISABLE_COMPOSITING_MODE=1

snhirsch, 2024-09-20:
Interesting!  I spun up a VMware guest running Debian Bookworm and the Connect tab appears to be working correctly in that environment.  I did note that Prusa is distributing (2) different versions of the slicer and it appears that Ubuntu 24.04 requires a different build from what I used.  The "Ubuntu" build appears to require a newer GLIBC version than Bookworm.

ahdinosaur, 2024-10-01:
Hi, I can confirm on Debian "bookworm", using proprietary Nvidia graphics drivers, using PrusaSlicer installed via flatpak, the window login and connect tab are blank. Fixed using either WEBKIT_DISABLE_DMABUF_RENDERER=1 or WEBKIT_DISABLE_COMPOSITING_MODE=1. Thanks for the help!

snhirsch, 2024-10-05:
Neither of those parameters work on Ubuntu 18.04 LTS.  I'm planning to upgrade my base install to Debian Bookworm in the near future and brought it up on a VM for testing.  So far it does seem to work properly in that environment.

miquelbeltran, 2024-10-18:
For anyone using flatpak, you can set the environment variable the setting permanently with:

```
sudo flatpak override --env=WEBKIT_DISABLE_COMPOSITING_MODE=1 com.prusa3d.PrusaSlicer
```

Then you can launch the app as usual. You can run the same command with `--unset-env=WEBKIT_DISABLE_COMPOSITING_MODE` when you don't need this anymore.

snhirsch, 2024-10-19:
I've found that the environment setting is required on Ubuntu 24.04 but not on the upstream Debian Bookworm.  Probably down to the version of Webkit.  It has never worked on Ubuntu 18.04, but that's getting long in tooth.

BallieBall, 2024-10-19:
> For anyone using flatpak, you can set the environment variable the setting permanently with:
>
> ```
> sudo flatpak override --env=WEBKIT_DISABLE_COMPOSITING_MODE=1 com.prusa3d.PrusaSlicer
> ```
>
> Then you [...]

This worked perfectly for me, thank you

guiweber, 2024-11-18:
I can confirm that on CachyOS (Arch Linux) using proprietary Nvidia graphics drivers, using either WEBKIT_DISABLE_DMABUF_RENDERER=1 or WEBKIT_DISABLE_COMPOSITING_MODE=1 fixes the issue.

lukasmatena, 2024-11-21:
Can you guys please retest this in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1)? Thanks.

miquelbeltran, 2024-11-21:
> Can you guys please retest this in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1)? Thanks.

I can confirm it works with the alpha1 as available on flatpak beta, I have also disabled the env var:

```
flatpak remote-add --if-not-exists flathub-beta https://flathub.org/beta-repo/flathub-beta.flatpakrepo
flatpak install flathub-beta com.prusa3d.PrusaSlicer
sudo flatpak override --unset-env=WEBKIT_DISABLE_COMPOSITING_MODE
flatpak run com.prusa3d.PrusaSlicer//beta
```

[image]

lukasmatena, 2024-11-21:
@miquelbeltran Thanks for getting back. Do you need to set the env var explicitly? It should no longer be needed with 2.9.0-alpha1.

miquelbeltran, 2024-11-21:
> @miquelbeltran Thanks for getting back. Do you need to set the env var explicitly? It should no longer be needed with 2.9.0-alpha1.

Nope, I have already removed it from my flatpak setup, to ensure I ran `--reset` to remove any env overrides:

```
sudo flatpak override --reset 
```

Running `--show` returns empty:

```
sudo flatpak override --show
```

probonopd, 2024-12-08:
To everyone affected, please try the AppImage linked in

* https://github.com/prusa3d/PrusaSlicer/issues/13653

and please let us know how how it works for you. Thanks!

MindlessDupe, 2024-12-13:
> That is not my issue. **I cannot enter the login password into the slicer at all, as the dialog box pops up blank and never fills in.** The same happens with the Physical Printer tab. My guess is [...]

Same/Similar issue here: I cannot log into my account in PrusaSlicer. Ver. 2.8.1 and Windows 11. When I open Slicer and click on "Log in" nothing happens, then I get the spinning circle cursor for many minutes. Clicking on the screen gets a fogged over look. X'ing out of the program gets the standard "Close" or "Wait" option when a program is hanging. <sadness>

idavydov, 2025-01-05:
I'm still having this problem on v2.9.0 installed via flatpak.

I also have the following environmental variables set:

```
$ flatpak override --show com.prusa3d.PrusaSlicer
[Environment]
WEBKIT_DISABLE_COMPOSITING_MODE=1
WEBKIT_DISABLE_DMABUF_RENDERER=1
```

Not sure if it's relevant, but in cases where I do see the printers sometimes I see a blank page sometimes for a half a second. It looks as if the page redirects somewhere.

idavydov, 2025-01-05:
One more thing to add, it seems to be related to notifications in the right bottom corner of the screen.

[image]
 
If I close all of them when the slicer starts the "Choose a printer" window mostly works fine. If I do not close them I often have an empty window.

41chemist, 2025-03-21:
I'm having the exact same issue. "choose a printer" comes up blank no matter what I do. I can't believe this issue was opened 9 months ago with dozens of people commenting on it and it still isn't fixed.

I'm running Ubuntu 24.04.2 LTS on a framework 16. if there's anything I can do to fix this, I'd be happy to provide files and test as much as needed.

SachCZ, 2025-03-21:
Hello @41chemist, can you please confirm what version of PrusaSlicer version are you using? Also, as you are willing to help, did you try this https://github.com/prusa3d/PrusaSlicer/issues/12969#issuecomment-2482106239?

Thank you in advance.
This issue is very hard to debug/fix, because we have very hard time reproducing it/identifying the problem.

41chemist, 2025-03-22:
@SachCZ Absolutely! I'm using the latest update (2.9.1), the stable release.

as to your question, I'm not using Nvidia graphics drivers. I'm using a framework 16 with a 7840HS and Radeon 780M discrete graphics (although I've tried temporary disabling it and using integrated graphics on my laptop to see if it fixed the issue. it did not)

Thank you so much for helping.

kocikdav, 2025-03-24:
hello @41chemist , since you are not using Nvidia graphics drivers, could you please try unsetting the WEBKIT_DISABLE_COMPOSITING_MODE and WEBKIT_DISABLE_DMABUF_RENDERER variables. Either by setting them to 0 or unsetting as @SachCZ proposed.

If these wont work: Your original comment says "choose a printer" comes out blank. So you are able to login without troubles? How about Printables and Prusa Connect tabs?

And just to confirm, you are using flatpack, right?

closed 2026-09-29 by prusaslicer-bot[bot]

mw46d, 2026-10-12:
Good news, not really:-( I now see the same issue on Kubuntu 24.04 with Prusa-Slicer 2.9.6 & 3.0.0.-alpha12 :-(

[image]