prusa3d/PrusaSlicer #12970: No longer saves as stl file

SteffJay, 2024-06-29:
### Description of the bug

After splicing, i can no longer save as an stl file (g-code). It only saves as a bg-code.

### Project file & How to reproduce

[axle_0.4n_0.2mm_PETG_MK3.5_35m.zip](https://github.com/user-attachments/files/16042182/axle_0.4n_0.2mm_PETG_MK3.5_35m.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

win64 v2.8.0

### Operating system

Windows 10 64bit

### Printer model

Prusa v3.5

Barnabasek, 2024-06-29:
Make sure you have unchecked the box "Use binary G-code when the printer supports it" if you want .gcode and not .bgcode
(in Menu - Configuration - Preferences - Other)

[image]

SteffJay, 2024-06-30:
All sorted, thank you.

kubispe1, 2024-07-02:
Or you eventaully can switch it off in Printer settings:
[image]
Closing the issue
See you

closed 2024-07-02 by kubispe1